nmikhailov/Validity90 #2: Can this driver also work on 0094 device?

ammen99, 2017-02-07:
Hello, I have a validity 0094 fingerprint sensor, please let me know if the driver you work on might be able to work on my device(I've seen that for ex. 0011 and 0018 sensors have the same driver). I can help with testing should you need it.

nmikhailov, 2017-02-07:
I have no idea. Maybe it would just work, maybe its significantly different. I will see how can I help you after I will finish with 0090. Event if its not significantly different you would most likely need to install win10 VM, sniff traffic and debug driver.

nmikhailov, 2017-04-23:
@ammen99 you can try to test https://github.com/nmikhailov/Validity90/tree/master/prototype , but most likely that won't work.
Will you be able to sniff usb traffic from win10 VM with driver installed?

maz3max, 2017-04-23:
Hi, I got a similar device with pid 97.
I got a Windows 10 VM in Virtualbox, how can I sniff USB traffic?

nmikhailov, 2017-04-23:
I use wireshark for that. You can run it in on host or in VM.

nmikhailov, 2017-04-23:
Also please provide links to windows drivers.

maz3max, 2017-04-23:
I just ran the prototype and got some output:
`Expected at char 012
Expected:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 26 85 88 42 45 3b  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 07 
Got:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 76 cf 48 9b 45 65  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 07 
Expected len: 84, but got 108
Expected:
0000 00 00 01 00 00 00 06 00  98 c9 71 56 01 00 34 46 
0010 02 00 07 00 60 39 00 00  01 00 84 08 01 00 07 00 
0020 b0 02 00 00 02 00 84 28  03 00 12 00 10 10 00 00 
0030 02 00 66 37 01 00 0c 00  40 22 02 00 01 00 86 47 
0040 00 00 01 00 20 5a 00 00  02 00 23 77 00 00 01 00 
0050 d0 2f 00 00 
Got:
0000 00 00 01 00 01 00 08 00  dd a6 4e 57 01 00 34 46 
0010 02 00 07 00 60 39 00 00  01 00 84 08 01 00 07 00 
0020 00 04 00 00 02 00 84 28  03 00 12 00 e0 10 00 00 
0030 02 00 76 36 01 00 0c 00  10 0a 00 00 01 00 86 47 
0040 00 00 01 00 50 5a 00 00  02 00 23 77 00 00 01 00 
0050 80 2f 00 00 02 00 66 37  01 00 0c 00 f0 22 02 00 
0060 01 00 25 56 00 00 01 00  70 04 00 00 
Expected at char 000
Expected:
0000 00 00 
Got:
0000 be 04 
Expected at char 031
Expected:
0000 00 00 00 10 00 00 00 00  00 00 01 00 6e 34 0b 9c 
0010 ff b3 7a 98 9c a5 44 e6  bb 78 0a 2c 78 90 1d 3f 
0020 b3 37 38 76 85 11 a3 06  17 af a0 1d 00 04 00 a1 
0030 00 c8 38 d8 e1 db f5 04  53 04 1a c5 a7 b4 0b 2f 
0040 1e f2 7d 7e 1b fd 48 da  a9 42 06 59 f3 3b 07 a7 
0050 e3 02 65 4c 1a dd a3 57  65 13 84 c7 98 38 4e 5e 
0060 d9 c7 33 5c ed 15 55 3c  f5 f4 de 14 a0 f2 59 68 
0070 00 a2 a0 98 58 c2 06 67  d5 c1 06 e3 bf e6 6a ec 
0080 6a c0 2d b2 d8 77 d9 0e  c4 12 e3 ab 48 ab aa b4 
0090 b9 56 75 30 69 9d 0a c3  d9 bb ff de 42 11 bd 34 
00a0 03 21 cf a2 8d 3c 1b e4  ba f0 1f f4 40 69 6f b4 
00b0 78 18 f3 2d 6b 22 80 86  64 31 14 34 2a 81 2c cc 
00c0 d7 c6 62 f3 9e 5f 78 a6  39 d3 db 57 c3 30 d4 dd 
00d0 12 8f 12 90 7e 4b 95 09  0e fa a2 e3 17 07 e9 74 
00e0 d8 33 a2 42 20 00 9a 33  ca 70 1c b9 3f 02 6e 78 
00f0 a2 ca 03 00 b8 00 ed 52  bb 71 b3 d9 0c 00 86 ad 
0100 64 0d 45 76 c7 32 b6 d5  d3 39 2d 89 5e 65 4b 60 
0110 6a 82 6a e5 bd 0c 17 00  00 00 20 00 00 00 ab 9d 
0120 fd ba 74 25 29 93 9d 2d  5d f4 77 ec 90 2e 13 b8 
0130 21 1a 19 70 1e 50 2f f5  6e 6e 25 ae 8c 00 00 00 
0140 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0150 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0160 00 00 dd f4 04 74 f0 7a  e4 e0 79 d1 f1 9f ae bd 
0170 a8 ef 1e fa 18 c2 6a 76  ae a5 aa bf c3 4f 12 94 
0180 8c 8f 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0190 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
01a0 00 00 00 00 00 00 00 00  00 00 00 00 00 00 a5 58 
01b0 ed 0f 31 33 45 63 c8 8a  d5 53 d9 e4 6e 20 5d 54 
01c0 3b 83 99 cf 9b ef 9e a8  aa c5 eb fb 20 a2 05 00 
01d0 a4 01 ec 5d 90 0e 5a 79  58 6d 2c db ee c6 22 40 
01e0 c6 89 9d 37 47 5e 0f 46  bb 9e fd 3f 5a 4f 32 e8 
01f0 27 d2 17 00 00 00 00 01  00 00 01 00 00 00 fc ff 
0200 ff ff ff ff ff ff ff ff  ff ff 00 00 00 00 00 00 
0210 00 00 00 00 00 00 01 00  00 00 ff ff ff ff 00 00 
0220 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0230 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0240 00 00 4b 60 d2 27 3e 3c  ce 3b f6 b0 53 cc b0 06 
[... 574 lines cut ...]
0010 1b 7e 50 c4 41 f7 74 fe  59 f1 a5 66 46 2f 14 9b 
0020 cd e3 7f 76 6c 02 21 00  ec 4e a0 55 86 26 f1 8a 
0030 f7 20 6c 02 98 a8 8f 5e  bb 30 c7 74 ab 21 26 aa 
0040 76 93 ab 93 5d 76 92 00 
HASHING>>>
0000 0f 00 00 48 30 46 02 21  00 b3 d5 bc 54 be 61 0b 
0010 1b 7c ff 3c 1b 7e 50 c4  41 f7 74 fe 59 f1 a5 66 
0020 46 2f 14 9b cd e3 7f 76  6c 02 21 00 ec 4e a0 55 
0030 86 26 f1 8a f7 20 6c 02  98 a8 8f 5e bb 30 c7 74 
0040 ab 21 26 aa 76 93 ab 93  5d 76 92 00 
HASHING<<
hash of handshake messages
0000 1a 24 6d 00 83 9d 23 26  f5 aa d2 d9 c4 30 63 52 
0010 f1 11 1e 58 d4 f9 2c 64  48 ef 4c 6b 46 df db b5 
client finished
0000 14 00 00 0c 36 5e b8 a8  f9 ee ab 56 aa 4b 24 1f 
final
0000 4b 77 62 ff a9 03 c1 1e  6f d8 35 93 17 2d 54 ef 
0010 96 ff 5f 5b b6 bd 6d bb  7e aa 3a 51 26 85 42 00 
0020 6c 7c ce c7 f0 b4 f4 6b  72 4b 0d a7 b6 cc de 33 
0030 d4 0e f9 6e ce 30 f1 a3  9c e2 7e 25 0f 20 6c 93 
0040 41 22 64 b0 52 97 7f 6d  c6 69 bc e1 e1 a0 03 56 
IN: 37b50a1e6d3ff33f7565ff747630d610041e4f9343ca8e0bd060baaed8db9081
OUT: 06f79ed900b755ee4dd85e1f5c32c83233f4def0ede4d95e9578f0f0d1d98eee
Failed 'EVP_DecryptUpdate(context, res, &tlen1, buff + 0x10, res_len)': 0 - LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED`

ammen99, 2017-06-02:
I tested it on my 0094 device, the program crashes with the following error:
`139622513104896:error:1007C06B:elliptic curve routines:EC_POINT_set_affine_coordinates_GFp:point is not on curve:crypto/ec/ec_lib.c:707:`

Here is the full output: [https://pastebin.com/FMyWqTYy](url)
Also, when I open wireshark, I get usbmon1/2/3, how do I understand which one I should listen on?

nmikhailov, 2017-06-02:
Either user usbmon0 to capture on all buses, or usbmon<bus_number_of_sensor>

Better write such reports to gitter - https://gitter.im/Validity90/Lobby

nmikhailov, 2017-06-13:
@ammen99 @maz3max Test the new version and report in our gitter - https://gitter.im/Validity90/Lobby

closed 2017-06-13 by nmikhailov