nmikhailov/Validity90 #3: Prototype not working

Leseratte10, 2017-05-18:
I also have a Validity 0090 finger print reader and I tried the prototype, but it prints errors. In case the output log of the prototype helps you to improve the prototype / driver, here it is: 

`usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 bb 2b 08 9b 45 56  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 02 
Expected at char 012
Expected:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 26 85 88 42 45 3b  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 07 
Got:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 bb 2b 08 9b 45 56  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 02 
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 67 27 1b c5  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
usb write:
0000 43 02 
usb read:
0000 af 04 
Expected len: 84, but got 2
Expected:
0000 00 00 01 00 00 00 06 00  98 c9 71 56 01 00 34 46 
0010 02 00 07 00 60 39 00 00  01 00 84 08 01 00 07 00 
0020 b0 02 00 00 02 00 84 28  03 00 12 00 10 10 00 00 
0030 02 00 66 37 01 00 0c 00  40 22 02 00 01 00 86 47 
0040 00 00 01 00 20 5a 00 00  02 00 23 77 00 00 01 00 
0050 d0 2f 00 00 
Got:
0000 af 04 
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6 
[... 353 lines cut ...]
0100 bf c5 9b 3c 67 d9 45 06  38 da 92 be 65 bf 81 8c 
0110 aa 7e 0f 00 00 48 30 46  02 21 00 fc 70 9f 00 5b 
0120 ad dc c5 58 69 b4 af 93  e9 1d 40 dd 7c 7d e1 35 
0130 4c 39 5e 78 59 6a cc 47  e3 43 55 02 21 00 92 e3 
0140 04 f8 fd d2 76 0a 06 43  13 43 9f bd 5a b3 89 ed 
0150 a1 e0 31 8d 35 52 28 a9  02 77 08 c3 0f 01 14 03 
0160 03 00 01 01 16 03 03 00  50 4b 77 62 ff a9 03 c1 
0170 1e 6f d8 35 93 17 2d 54  ef 90 56 a6 9f f7 69 6c 
0180 3d 54 b2 18 a3 9f c6 fc  a3 8a b2 e3 cf b3 cc a2 
0190 80 e8 b2 bd a2 41 0e ba  1f 26 43 34 4e de 56 63 
01a0 b6 d9 ba 90 53 2f ce 2c  15 a5 d8 a5 32 39 32 1b 
01b0 aa 2b a2 ce ce 5d f4 7d  53 
usb read:
0000 b1 04 
IN: c7fc0d98f59cb84a85710d49412087d984518b7100d8ceb41ab280fa29d32ac6
OUT: 7d121e2a031ab6db3666b2215e06cbe514376029ab45a1780d4b2ba5edb0067a
usb write:
0000 17 03 03 00 40 4b 77 62  ff a9 03 c1 1e 6f d8 35 
0010 93 17 2d 54 ef ae 86 3f  af 53 82 47 f9 f9 05 1c 
0020 97 72 27 5f 34 09 99 f7  1d cb ca 31 d7 97 3c cc 
0030 05 77 10 79 50 8d 69 23  e9 0b fa ef c9 07 30 8f 
0040 91 27 5f 6f 70 
usb read:
0000 00 00 00 00 00 00 00 
Failed 'EVP_DecryptUpdate(context, res, &tlen1, buff + 0x10, res_len)': 0 - LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED`

lindhe, 2017-05-18:
I'd recommend checking out the gitter: https://gitter.im/Validity90/Lobby

There's a discussion related to this.

nmikhailov, 2017-06-01:
Yep, @lindhe is right, its better to discuss this on gitter, also there is a new version available so you can retest.

closed 2017-06-01 by nmikhailov