nmikhailov/Validity90 #4: Prototype output on T470p

wjdp, 2017-08-05:
```
Prototype version 10b
Found device 138a:0097
This device is not supported, but lets try anyway
Index 1, size 12
0000 37 37 33 32 38 30 62 39  31 34 64 36 

usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 77 23 08 9b 41 6d  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 07 
Expected at char 012
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 ef 87 51 b1  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
usb write:
0000 43 02 
usb read:
0000 00 00 01 00 01 00 08 00  dd a6 4e 57 01 00 34 46 
0010 02 00 07 00 60 39 00 00  01 00 84 08 01 00 07 00 
0020 00 04 00 00 02 00 84 28  03 00 12 00 e0 10 00 00 
0030 02 00 76 36 01 00 0c 00  10 0a 00 00 01 00 86 47 
0040 00 00 01 00 50 5a 00 00  02 00 23 77 00 00 01 00 
0050 80 2f 00 00 02 00 66 37  01 00 0c 00 f0 22 02 00 
0060 01 00 25 56 00 00 01 00  70 04 00 00 
Expected len: 84, but got 108
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6 
01e0 31 28 dc 82 45 
usb read:
0000 be 04 
Expected at char 000
usb write:
0000 3e 
[... 971 lines cut ...]
usb read:
0000 17 03 03 00 40 c5 49 e1  d8 8b 87 db 36 d5 ca d6 
0010 91 f9 08 7b f0 ef 4b 1b  68 be 22 1a 10 83 78 d5 
0020 81 4d e7 0c 9f 9c 3d ac  ee f9 2a d2 49 f1 5d 67 
0030 51 70 d8 e4 70 91 6d d3  af 6b 85 94 99 64 1b b4 
0040 b7 c6 c8 60 ce 
usb write:
0000 17 03 03 00 40 4b 77 62  ff a9 03 c1 1e 6f d8 35 
0010 93 17 2d 54 ef b2 27 58  20 86 4a bf a7 fd cf c7 
0020 66 fa a3 41 c1 e7 3b 2e  ae fc 70 e7 91 05 58 37 
0030 e8 b5 c1 9c 33 31 cd 57  85 b5 f4 80 0e 2f f5 2f 
0040 3a 26 f4 2d 9d 
usb read:
0000 17 03 03 00 40 04 db db  96 8c d4 29 67 90 23 4a 
0010 a3 bf 59 64 d6 3b e7 11  2c e3 1b c7 e5 56 67 ea 
0020 9b a2 d5 92 61 60 f3 38  b3 42 42 ff 66 0f 27 cf 
0030 65 51 34 35 fc 7d 5d 94  2c d9 55 63 13 7a 06 5c 
0040 bc 8a 17 d1 94 
total len  0
Image written - img.png, img.raw
```

[image]

Anything else I can help with, let me know.

kowalk, 2017-09-10:
Hi, I'm attaching output from mine T470p, if you need something more then let me know.

[output.txt](https://github.com/nmikhailov/Validity90/files/1290846/output.txt)
[image]

nmikhailov, 2018-02-12:
Thanks. Prototype was updated, it doesn't output images anymore since they are not available on 0097. Hovewer it can now check fingerprints againts internal db. Please join our [gitter](https://gitter.im/Validity90/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link) to stay in touch and receive project updates.

closed 2018-02-12 by nmikhailov

wjdp, 2018-02-15:
Are you still wanting prototype output (i.e. should I get the new version and post?)

nmikhailov, 2018-02-16:
You should definetly post logs if new version doesn't work, otherwise logs are not needed.