nmikhailov/Validity90 #6: Fails to build

zippy2, 2017-08-27:
Maybe I'm doing something wrong, but all I did was:

clone the repo,
cd repo/libfprint && ./autogen.sh && make

and I see a lot of compile errors, for instance:

drivers/vfs0090.c: In function ‘async_write’:
drivers/vfs0090.c:82:46: error: ‘VFS_USB_TIMEOUT’ undeclared (first use in this function)
                   async_write_callback, ssm, VFS_USB_TIMEOUT);
                                              ^
drivers/vfs0090.c:82:46: note: each undeclared identifier is reported only once for each function it appears in
drivers/vfs0090.c: In function ‘async_read’:
drivers/vfs0090.c:127:15: error: ‘EP3_IN’ undeclared (first use in this function)
     if (ep == EP3_IN)
               ^
drivers/vfs0090.c:130:28: error: ‘VFS_USB_TIMEOUT’ undeclared (first use in this function)
                            VFS_USB_TIMEOUT);
                            ^

And the list goes on and on. Also, any plans on merging this to libgprint later on? Thanks.

ammen99, 2017-08-27:
I'm not the developer of this driver, but I believe that the libfprint driver isn't ready for use(or even for compilation). If you want to test, try the prototype.

zippy2, 2017-08-27:
Oh, that one fails to initialize the device:

validity90.git/prototype # ./prototype | grep "Expected"
Expected at char 012
Expected len: 84, but got 2
Expected len: 76, but got 16
Expected len: 4104, but got 2

validity90.git/prototype # ./prototype | tail -n5
0000 40 01 01 00 00 00 00 00  00 00 10 00 00 
usb read:
0000 af 04 
Expected len: 4104, but got 2
Invalid rsp6, can't get public key

Danyc0, 2017-08-28:
What reader are you using? Looks like it might not be supported by this project

zippy2, 2017-08-28:
Bus 001 Device 003: ID 138a:0090 Validity Sensors, Inc.

lindhe, 2017-08-28:
Please correct me if I'm wrong, but since the driver is not ready yet I don't think you should be able to build that yet. If you go to validity90.git/prototype instead and `make`, does that work? Because I think that's all that is supposed to work so far. But I might be wrong.

zippy2, 2017-08-28:
Well, as I say in the 3rd comment: libfprint provided with this repo fails to build; and the prototype does build but fails to initialize the device. Unsure why.

lindhe, 2017-08-28:
Ah. Do you run the prototype as root (`sudo ./prototype`)?

zippy2, 2017-08-28:
Yes. If you read the comment carefully you'll notice the sharp sign.

lindhe, 2017-08-28:
I saw that, but everyone has a different prompt.

ammen99, 2017-08-28:
@zippy2 Have you used the device with windows? If not, try to install it in a VM(if you don't want to install it on hardware), setup your fingerprint and try prototype again