nmikhailov/Validity90 #8: Prototype output T470 (Not supported 138a:0097)

Eik-S, 2017-08-30:
`
Prototype version 10b
Found device 138a:0097
This device is not supported, but lets try anyway
Index 1, size 12
0000 62 61 39 33 38 34 65 64  31 34 65 38 

usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 ab 39 48 de 41 8e  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 02 
Expected at char 012
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 85 93 39 cc  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
usb write:
0000 43 02 
usb read:
0000 af 04 
Expected len: 84, but got 2
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6 
01e0 31 28 dc 82 45 
usb read:
0000 be 04 
Expected at char 000
usb write:
0000 3e 
usb read:
0000 00 00 ef 00 40 00 00 10  10 00 00 01 01 00 00 00 
Expected len: 76, but got 16
usb write:
0000 40 01 01 00 00 00 00 00  00 00 10 00 00 
usb read:
0000 af 04 
Expected len: 4104, but got 2
Invalid rsp6, can't get public key
`

Eik-S, 2017-08-30:
maybe helps for supporting it ;)

HerHde, 2017-09-05:
Mine is a T570, but same sensor
https://gist.github.com/HerHde/6417d7709b35f54823f343ecaea02fe5

erdenezul, 2018-03-13:
Green on
usb write:
0000 17 03 03 00 b0 4b 77 62  ff a9 03 c1 1e 6f d8 35 
0010 93 17 2d 54 ef 63 31 a9  2a c2 5f 37 1c 31 57 b7 
0020 8f c2 69 f9 2b 14 84 b4  b3 99 bf 13 44 98 54 1e 
0030 1d b2 99 1b 64 c4 cb cb  8c 24 2f 40 02 fb 9e 2f 
0040 82 f7 70 97 36 f3 83 68  04 d1 91 a0 bb 12 e9 19 
0050 15 ff 0d d5 bc 7f 64 f9  ca 5a 77 d4 dd e4 d8 ed 
0060 ec e0 eb 59 01 e0 a3 3a  ff 7d 79 06 4d 22 29 37 
0070 f5 34 51 97 9c 42 91 17  49 50 33 d0 81 eb 63 e2 
0080 26 c1 aa a0 31 6c c0 b2  8c 75 14 54 89 a1 bf 4d 
0090 94 5d 22 6f 11 0d 6a 6a  51 15 80 95 5c 30 e9 2c 
00a0 7b 18 93 f5 c3 55 87 cb  d2 e1 50 fa 0c da 98 d6 
00b0 4e 3c 9a 5d 83 
usb read:
0000 17 03 03 00 40 cf 9d 5d  08 2a bd 56 45 63 2d b5 
0010 6d d7 ab b5 57 0d b3 8a  0d ed 82 b6 df 11 4f 5d 
0020 f8 0c 18 d1 89 c2 80 ef  cc f1 82 92 f8 cf 06 53 
0030 85 ba 92 45 2d f6 6a ec  fc 18 88 a9 54 14 2a 71 
0040 ec 8d 9c 85 a7 
READ:
0000 00 00 
Red blink x3 then off
usb write:
0000 17 03 03 00 b0 4b 77 62  ff a9 03 c1 1e 6f d8 35 
0010 93 17 2d 54 ef 66 9f 22  d0 53 f6 b9 eb 09 28 2e 
0020 14 ec b1 df 1e 60 46 50  98 a5 4f 7a ca 95 8e 9b 
0030 aa be 29 97 f3 34 2b cc  a7 e1 e5 06 64 2d 18 43 
0040 b4 ec 25 37 9a 56 a0 e3  b0 ff bb b9 c2 94 1a 2c 
0050 ba 08 6f 63 e4 4f 4c ff  e3 c6 53 c1 a1 24 9c ac 
0060 87 f1 db e5 78 e5 ba c6  84 10 11 b9 3c b5 4a a3 
0070 55 d9 97 6e cd 7b a4 f6  6d e4 e1 85 21 f2 09 13 
0080 6e 3f 34 97 32 bc 2a 92  ee 5c 1e 7b 80 a1 6c a9 
0090 65 60 02 7e 91 c4 c0 42  cc 10 bb df ad a5 b9 67 
00a0 2d b4 58 96 b9 5e fa 32  86 b4 88 19 12 ba 7f 7b 
00b0 de 5c b3 a7 d9 
usb read:
0000 17 03 03 00 40 50 f7 8f  05 5d 74 09 df f5 6d 92 
0010 bc a7 bc d8 a9 2c 95 47  89 30 f0 76 b3 ca b5 b1 
0020 95 b4 fd 8c 71 10 fa b0  f7 3c ca 7f d3 6c f7 ca 
0030 2f 02 c0 87 73 60 ce 9a  62 3c 91 e3 67 26 c8 31 
0040 9f a0 9c 77 78 
READ:
0000 00 00 
Green blink
usb write:
0000 17 03 03 00 b0 4b 77 62  ff a9 03 c1 1e 6f d8 35 
0010 93 17 2d 54 ef 1f 45 81  fb 00 23 c5 15 66 8a ea 
0020 99 41 d0 34 58 5c fd d3  b7 8b 62 5e 19 96 96 ce 
0030 74 d1 29 df 96 c0 9c fb  27 ba 16 72 9f cf 14 76 
0040 ae 34 7a 25 f9 16 22 e7  b2 b5 6d 00 64 b0 66 ac 
0050 37 80 68 31 01 94 1e 64  a2 37 df 60 e1 39 54 4b 
0060 d4 b1 a8 dc e9 4b f2 88  35 46 fe 50 32 d0 0f 96 
0070 98 b2 c4 95 a7 bf 5e 63  d7 c3 8c 41 7c 71 c1 b6 
0080 bc 58 f9 7e 86 e5 fd e6  8f 0c 26 63 eb 7a ec bd 
0090 1a de 48 21 84 5b d9 03  d8 92 db fc d3 a5 81 25 
00a0 67 98 2b 94 3b b9 1a 75  d8 78 f1 04 7b dc b5 d0 
00b0 91 dc b2 f9 90 
usb read:
0000 17 03 03 00 40 51 b2 72  01 5c ec ef 6b 45 66 e8 
0010 42 8c dc cb 2c b0 0e 1a  f3 7c 34 b5 d0 a2 aa 4a 
0020 1d 30 cf 6f 58 71 46 69  4d 0a 31 4a 91 67 70 21 
0030 a3 6c a3 7e 56 fa 81 24  17 53 bf 7e 6f b0 93 b2 
0040 48 2c 91 4f 14 
READ:
0000 00 00 
Custom script
usb write:
0000 17 03 03 00 b0 4b 77 62  ff a9 03 c1 1e 6f d8 35 
0010 93 17 2d 54 ef 47 bb f4  d3 f2 f0 d8 4e 60 19 80 
0020 7e 0c 16 de c5 92 14 2e  2a 50 9e e4 c6 09 f8 55 
0030 1f a5 6b 93 03 ce bd 8b  e8 6a 6e 10 11 8d 21 91 
0040 a0 ae 33 4e 4a e0 52 0a  0d 47 89 77 c9 76 18 fa 
0050 f1 a4 78 ef 96 69 47 ef  21 06 d8 18 05 67 04 d0 
0060 3a 61 46 37 dd b6 eb af  a9 82 6d e7 b4 42 9d 30 
0070 13 ec 0c a5 17 22 40 7c  36 d5 80 ab ab 24 83 86 
0080 ac 07 a1 8c d5 9d 55 be  77 a9 9e 3d dc 86 02 3d 
0090 9c 6f d2 85 ac 49 bc e3  75 33 b5 05 2b 2e 64 01 
00a0 39 80 d2 e5 3b da de 60  5d 6f 21 e7 ee 6f 8a 5e 
00b0 fa aa c9 2f 70 
usb read:
0000 17 03 03 00 40 42 2a e6  ba 23 d6 31 b6 a3 d9 1f 
0010 b1 c0 e2 02 9d b2 2b da  4a 45 4c 17 4f a1 d8 1e 
0020 c0 55 b0 46 f1 f1 6f c8  66 f8 a5 cb 51 d9 52 ad 
0030 da 4e d0 d8 6a 06 c9 d0  34 e0 e3 6e 2a 83 68 71 
0040 53 89 32 1d ca 
READ:
0000 00 00

erdenezul, 2018-03-13:
looking forward to support 0097 driver to work on ubuntu.

benediktg, 2018-04-25:
Here is my output (Thinkpad T470):
<details>
<summary>Output</summary>

```
Prototype version 15
Found device 138a:0097

step 1
usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 fb cb 08 98 45 94  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 02 
Expected at char 012
Sensor not initialized, init byte is 0x2 (expected 0x02)
step 2
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 5e 31 18 bc  00 00 00 00 98 ec 72 e2 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 5e 31 18 bc  00 00 00 00 98 ec 72 e2 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
step 4
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6 
01e0 31 28 dc 82 45 
usb read:
0000 be 04 
Expected at char 000
0000 be 04 
step 5
usb write:
[... 808 lines cut ...]
0000 01 04 
Expected len: 298, but got 2
0000 01 04 
step 12
usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 fb cb 08 98 45 94  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 02 
Expected at char 011
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 fb cb 08 98 45 94  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 02 
step 13
usb write:
0000 1a 
usb read:
0000 91 04 
Expected at char 000
0000 91 04 
Incomplete reverse engineering
```

</details>