nmikhailov/Validity90 #10: "This device doesn't use encryption and is currently out of scope for this project"

thecosmicfrog, 2017-09-14:
Hi @nmikhailov,

Thanks for your work on this project. I decided to try the prototype on a Dell XPS 13 (9360) [2017 model] running Ubuntu 17.04. I get the below output:

    Prototype version 12
    Found device 138a:0091
    This device is UNsupported. This device doesn't use encryption and is currently out of scope for this project

When you say "doesn't use encryption", in what context does this mean? The storage of the fingerprint itself is not encrypted or it's not encrypted on-the-wire? Or something else?

Are those of us with Dell XPS 13s out of luck completely?

Thanks again,
Aaron

nmikhailov, 2017-09-15:
91's protocol is different from 90, 94, 97. Its much simpler because communications between driver and sensor are not encrypted, I was even able to extract fingerprint image from the traffic dump.

mind-blowing, 2017-09-15:
Hi @nmikhailov ,

I'm waiting for my Dell XPS 15 (9560) and based on issue #5 it has same device as on Dell XPS 13 (9360), can you please clarify some unclear moment, does your answer means that you are not going to support this device at all or priority for supporting of this device is lower then for other (90, 94, 97) ?

Thank you for your job,
Ivan

nmikhailov, 2017-09-15:
Right now it looks very unlikely. I might consider supporting it(as it is now, it probably won't even share code with this project), but it would be in the distant future.

Even if I would want to support it, it would be impossible without this sensor itself. 
You can link me this part in taobao/aliexpress so I can check it its connector looks anything like what I have.

There is also a posibility that something would be discovered that would unify 91 with the rest (ie that others have fallback mode which works like 91).

Also you can provide your version of dumps, so we can confirm the difference.

Danyc0, 2017-09-15:
@nmikhailov: You say you were able to extract the fingerprint from a dump of a 91, could you briefly explain how you did that?

nmikhailov, 2017-09-16:
It is just there in the dump as roughly 150x150 pixel grayscale image.

ingmarcosp, 2018-01-01:
I found this image, if you want to look how is it.
https://img2.parts-people.com/products/M0T6Pc1.JPG

hmaarrfk, 2018-03-04:
I'm reverse engineering the fingerprint reader:
https://github.com/hmaarrfk/Validity91

You can tell me how this goes for all of you.

nmikhailov, 2018-03-04:
I have linked your repo in the status. Closing this as out of scope.

closed 2018-03-04 by nmikhailov