nmikhailov/Validity90 #9: Prototype Output T470s

tuxBurner, 2017-09-14:
Prototype version 12
Found device 138a:0097
This device support is in progress, it shouldn't print correct fingerprint image right now
Index 1, size 12
0000 61 31 64 63 38 38 35 33  35 34 39 36 

usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 1a cd 88 35 45 69  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 07 
Expected at char 012
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 3f 44 08 6f  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
usb write:
0000 43 02 
usb read:
0000 00 00 01 00 01 00 08 00  dd a6 4e 57 01 00 34 46 
0010 02 00 07 00 60 39 00 00  01 00 84 08 01 00 07 00 
0020 00 04 00 00 02 00 84 28  03 00 12 00 e0 10 00 00 
0030 02 00 76 36 01 00 0c 00  10 0a 00 00 01 00 86 47 
0040 00 00 01 00 50 5a 00 00  02 00 23 77 00 00 01 00 
0050 80 2f 00 00 02 00 66 37  01 00 0c 00 f0 22 02 00 
0060 01 00 25 56 00 00 01 00  70 04 00 00 
Expected len: 84, but got 108
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6 
01e0 31 28 dc 82 45 
usb read:
0000 be 04 
Expected at char 000
usb write:
0000 3e 
usb read:
[... 967 lines cut ...]
0020 1b 2b 2d 6e aa a7 ec a1  5a 6d d7 14 37 82 24 4f 
0030 c1 14 44 af ad b9 66 88  f2 91 3a 6b 1d 8c 2d 0c 
0040 0a eb 21 12 48 
usb read:
0000 17 03 03 00 40 4e aa 59  01 ee 61 d0 45 4e 48 3b 
0010 99 21 58 8e 88 3d ee ab  57 5b 03 c4 2a 11 9c ee 
0020 9c 29 65 87 98 b0 34 1c  bc c3 ba 08 1c 38 90 37 
0030 34 33 69 17 6b c9 2d 4d  2c 8b 75 63 9d c1 20 8e 
0040 16 d2 1e c3 1e 
usb write:
0000 17 03 03 00 40 4b 77 62  ff a9 03 c1 1e 6f d8 35 
0010 93 17 2d 54 ef 37 07 97  d3 9d 98 40 3d 2b 9f 8a 
0020 1b 2b 2d 6e aa a7 ec a1  5a 6d d7 14 37 82 24 4f 
0030 c1 14 44 af ad b9 66 88  f2 91 3a 6b 1d 8c 2d 0c 
0040 0a eb 21 12 48 
usb read:
0000 17 03 03 00 40 69 e2 f8  a4 62 bf b1 49 51 91 36 
0010 83 b9 41 49 06 ba 37 4b  51 1f c2 21 63 51 9b 4e 
0020 a4 db 64 0e af ac b1 53  82 ff 38 1a 79 53 a6 0b 
0030 f6 39 02 67 b4 04 42 86  db d8 73 d0 e0 a5 aa d2 
0040 23 25 a3 b3 0c 
total len  0
Image written - img.png, img.raw

[image]

tuxBurner, 2017-09-14:
Used the side of my finger :)

sesnik, 2017-11-24:
Here is the prototype output on my T470s

Prototype version 12
Found device 138a:0097
This device support is in progress, it shouldn't print correct fingerprint image right now
Index 1, size 12
0000 39 37 34 35 38 63 38 39  31 34 36 36 

usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 79 54 c8 98 41 66  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 07 
Expected at char 012
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 91 f0 ff 5d  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
usb write:
0000 43 02 
usb read:
0000 00 00 01 00 01 00 08 00  5e 78 2d 59 01 00 34 46 
0010 02 00 07 00 30 3e 00 00  01 00 84 08 01 00 07 00 
0020 00 04 00 00 02 00 84 28  03 00 12 00 e0 10 00 00 
0030 02 00 76 36 01 00 0c 00  10 0a 00 00 01 00 86 47 
0040 00 00 01 00 50 5a 00 00  02 00 23 77 00 00 01 00 
0050 80 2f 00 00 02 00 66 37  01 00 0c 00 f0 22 02 00 
0060 01 00 25 56 00 00 01 00  70 04 00 00 
Expected len: 84, but got 108
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6 
01e0 31 28 dc 82 45 
usb read:
0000 be 04 
Expected at char 000
usb write:
[... 971 lines cut ...]
0040 29 0d b0 50 a8 
usb read:
0000 17 03 03 00 40 d6 01 56  d9 63 31 05 7b a4 d5 06 
0010 e7 6c c5 72 34 a8 3d b7  06 58 85 42 4a 18 07 6e 
0020 64 34 c6 c4 4a 8a 51 8c  80 3f 0e 58 2d bd 4f 8c 
0030 1f 6d 0f 0b 25 e9 40 76  78 d2 ab 09 5e ff 15 52 
0040 ed 0f 5f c2 81 
usb write:
0000 17 03 03 00 40 4b 77 62  ff a9 03 c1 1e 6f d8 35 
0010 93 17 2d 54 ef 10 7b 3b  bc 1e 1e 4e 72 c3 66 9f 
0020 9e 25 05 87 56 d4 d4 c3  18 68 c2 d1 14 79 fd 8d 
0030 ad a4 9d 8b 50 fa e4 1a  76 f9 73 7c f2 c2 05 35 
0040 29 0d b0 50 a8 
usb read:
0000 17 03 03 00 40 6c 86 3f  d9 ca 5f 09 54 1c 69 1d 
0010 e8 9e ea d6 21 38 40 d1  4f e9 aa 5e 01 5a f5 a0 
0020 4f 6e c5 5e cd 4b 94 51  84 be 72 58 ca fb aa 23 
0030 d6 a0 2b bf ee d5 82 af  f5 9e e1 0a 43 63 e8 7a 
0040 02 d7 c8 07 c2 
total len  0
Image written - img.png, img.raw

[image]

Used fake finger data as well.

nmikhailov, 2018-02-12:
Thanks. Prototype was updated, it doesn't output images anymore since they are not available on 0097. Hovewer it can now check fingerprints againts internal db. Please join our [gitter](https://gitter.im/Validity90/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link) to stay in touch and receive project updates.

closed 2018-02-12 by nmikhailov