nmikhailov/Validity90 #11: Prototype Output ThinkPad Carbon X1 5th Gen

clstb, 2017-09-21:
ThinkPad Carbon X1 5th Gen
Arch Linux

Prototype version 12
Found device 138a:0097
This device support is in progress, it shouldn't print correct fingerprint image right now
Index 1, size 12
0000 31 34 64 66 38 30 38 39  31 34 38 39 

usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 41 fd 08 98 41 98  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 02 
Expected at char 012
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 5b 72 ff 9f  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
usb write:
0000 43 02 
usb read:
0000 af 04 
Expected len: 84, but got 2
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6 
01e0 31 28 dc 82 45 
usb read:
0000 be 04 
Expected at char 000
usb write:
0000 3e 
usb read:
0000 00 00 ef 00 40 00 00 10  10 00 00 01 01 00 00 00 
Expected len: 76, but got 16
usb write:
0000 40 01 01 00 00 00 00 00  00 00 10 00 00 
usb read:
0000 af 04 
Expected len: 4104, but got 2
Invalid rsp6, can't get public key

xvzf, 2017-10-01:
Prototype version 12
Found device 138a:0097
This device support is in progress, it shouldn't print correct fingerprint image right now
Index 1, size 12
0000 61 36 61 34 38 30 65 64  35 34 66 38 

usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 6a 4a 08 de 45 8f  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 07 
Expected at char 012
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 5c 2f e3 05  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
usb write:
0000 43 02 
usb read:
0000 00 00 01 00 01 00 08 00  dd a6 4e 57 01 00 34 46 
0010 02 00 07 00 60 39 00 00  01 00 84 08 01 00 07 00 
0020 00 04 00 00 02 00 84 28  03 00 12 00 e0 10 00 00 
0030 02 00 76 36 01 00 0c 00  10 0a 00 00 01 00 86 47 
0040 00 00 01 00 50 5a 00 00  02 00 23 77 00 00 01 00 
0050 80 2f 00 00 02 00 66 37  01 00 0c 00 f0 22 02 00 
0060 01 00 25 56 00 00 01 00  70 04 00 00 
Expected len: 84, but got 108
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6 
01e0 31 28 dc 82 45 
usb read:
0000 be 04 
Expected at char 000
usb write:
0000 3e 
usb read:
[... 965 lines cut ...]
0000 17 03 03 00 40 4b 77 62  ff a9 03 c1 1e 6f d8 35 
0010 93 17 2d 54 ef 17 c6 7b  fc bb ec 84 c9 d1 68 06 
0020 f7 93 01 94 2d 89 c4 a9  4b 39 a1 f8 ef a6 83 91 
0030 25 d2 52 cf 0f 14 3b 5f  78 dd 33 60 70 38 de 4d 
0040 4e b5 d2 3b 70 
usb read:
0000 17 03 03 00 40 0a 2f bc  98 01 89 4a f4 5f fb f1 
0010 6c 82 7e 4e 6a c2 7d 7c  7b e2 e5 80 b7 ea ca a6 
0020 91 8b e6 b9 83 c5 ee 48  62 48 12 40 53 71 0c 5c 
0030 7c 17 29 ed 21 a6 25 05  6e 37 92 46 d4 c7 14 f6 
0040 4c 03 3d ed 9b 
usb write:
0000 17 03 03 00 40 4b 77 62  ff a9 03 c1 1e 6f d8 35 
0010 93 17 2d 54 ef 17 c6 7b  fc bb ec 84 c9 d1 68 06 
0020 f7 93 01 94 2d 89 c4 a9  4b 39 a1 f8 ef a6 83 91 
0030 25 d2 52 cf 0f 14 3b 5f  78 dd 33 60 70 38 de 4d 
0040 4e b5 d2 3b 70 
usb read:
0000 17 03 03 00 40 a4 2a b4  c5 b8 c4 30 86 5e 0d bb 
0010 7f 1f c2 c9 d4 2c 1d d2  2a de 69 ca 70 74 de 6c 
0020 7b 49 fd 0f 02 95 47 ea  5c 0f 62 db 59 ff 6d fc 
0030 70 be 19 2b f5 16 ac e9  1d d3 94 80 86 ad e2 c9 
0040 8c 18 9c 7c 61 
total len  0
Image written - img.png, img.raw

nmikhailov, 2018-02-12:
Thanks. Prototype was updated, it doesn't output images anymore since they are not available on 0097. Hovewer it can now check fingerprints againts internal db. Please join our [gitter](https://gitter.im/Validity90/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link) to stay in touch and receive project updates.

closed 2018-02-12 by nmikhailov