nmikhailov/Validity90 #14: Prototype output for Lenovo T570

hinxx, 2017-10-01:
https://gist.github.com/hinxx/b769c09094184346f5ca168d9451331c

Thanks!

PofMagicfingers, 2017-10-03:
Another one : https://gist.github.com/PofMagicfingers/4799d8a3dc52926e68d44e8db0925539

Thanks !

pciavald, 2017-11-25:
https://gist.github.com/pciavald/611aad3153e0d3be74a4ff12cd40ff28

[image]

petrmanek, 2018-01-01:
Here's another one: https://gist.github.com/petrmanek/8d4c2d86e7c902b7d6185f922ee3e322
[image]

nmikhailov, 2018-02-12:
Thanks. Prototype was updated, it doesn't output images anymore since they are not available on 0097. Hovewer it can now check fingerprints againts internal db. Please join our [gitter](https://gitter.im/Validity90/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link) to stay in touch and receive project updates.

closed 2018-02-12 by nmikhailov