nmikhailov/Validity90 #16: Prototype output for 138a:0097 Validity Sensors, Inc - Thinkpad X270

sreed-dci, 2017-10-03:
Prototype version 12
Found device 138a:0097
This device support is in progress, it shouldn't print correct fingerprint image right now
Index 1, size 12
0000 64 31 36 35 38 30 65 64  35 34 61 39 

usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 1d 56 08 de 45 9a  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 07 
Expected at char 012
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 55 c6 2e e0  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
usb write:
0000 43 02 
usb read:
0000 00 00 01 00 01 00 08 00  dd a6 4e 57 01 00 34 46 
0010 02 00 07 00 60 39 00 00  01 00 84 08 01 00 07 00 
0020 00 04 00 00 02 00 84 28  03 00 12 00 e0 10 00 00 
0030 02 00 76 36 01 00 0c 00  10 0a 00 00 01 00 86 47 
0040 00 00 01 00 50 5a 00 00  02 00 23 77 00 00 01 00 
0050 80 2f 00 00 02 00 66 37  01 00 0c 00 f0 22 02 00 
0060 01 00 25 56 00 00 01 00  70 04 00 00 
Expected len: 84, but got 108
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6 
01e0 31 28 dc 82 45 
usb read:
0000 be 04 
Expected at char 000
usb write:
0000 3e 
usb read:
[... 966 lines cut ...]
0010 93 17 2d 54 ef ee 18 71  96 05 61 84 b2 4f 49 5f 
0020 b8 62 83 a7 cc 61 29 6d  e1 b7 19 18 0b b9 a3 90 
0030 b8 a6 70 87 56 00 10 0c  a9 2a 08 9a ea b0 9d f6 
0040 8a 54 cc ac 8b 
usb read:
0000 17 03 03 00 40 09 61 d9  fc 1a 32 36 7f 80 01 e0 
0010 a2 65 98 1c be e0 f0 94  96 b1 3b 46 ea 86 a4 9d 
0020 97 b9 fe 88 fc 38 0f 5a  68 4e 3f 6f 51 a9 0b c2 
0030 9d 11 54 6e 42 f1 2b 35  d4 b2 e9 a0 61 0d 5e eb 
0040 19 5c 15 7b d4 
usb write:
0000 17 03 03 00 40 4b 77 62  ff a9 03 c1 1e 6f d8 35 
0010 93 17 2d 54 ef ee 18 71  96 05 61 84 b2 4f 49 5f 
0020 b8 62 83 a7 cc 61 29 6d  e1 b7 19 18 0b b9 a3 90 
0030 b8 a6 70 87 56 00 10 0c  a9 2a 08 9a ea b0 9d f6 
0040 8a 54 cc ac 8b 
usb read:
0000 17 03 03 00 40 69 f5 a9  f9 9b 47 fb 13 49 a8 38 
0010 f3 4e f2 24 a1 52 9d ae  12 4d 3d 16 a6 85 cf 5d 
0020 19 3c e3 63 0f 0e 48 53  fc 7b 4a 2b 60 08 0d 2d 
0030 32 f2 c1 52 5f 15 59 43  cd 3c fe 69 fa e0 b8 5f 
0040 e4 6c b6 f5 c7 
total len  0
Image written - img.png, img.raw
[image]

lrizzi, 2017-10-05:
This is the output from another X270 I'm not sure why on this one the encryption is enabled:

`Prototype version 12
Found device 138a:0097
This device support is in progress, it shouldn't print correct fingerprint image right now
Index 1, size 12
0000 31 35 35 37 38 30 35 33  35 34 30 61 

usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 51 75 08 35 45 a0  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 02 
Expected at char 012
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 be 31 1f 97  00 00 00 00 8c c4 e2 57 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
usb write:
0000 43 02 
usb read:
0000 af 04 
Expected len: 84, but got 2
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6 
01e0 31 28 dc 82 45 
usb read:
0000 be 04 
Expected at char 000
usb write:
0000 3e 
usb read:
0000 00 00 ef 00 40 00 00 10  10 00 00 01 01 00 00 00 
Expected len: 76, but got 16
usb write:
0000 40 01 01 00 00 00 00 00  00 00 10 00 00 
usb read:
0000 af 04 
Expected len: 4104, but got 2
Invalid rsp6, can't get public key
`

timoludwig, 2018-02-14:
138a:0097 on ThinkPad X1-Carbon 5th 20HQS03P00
```
Prototype version 15
Found device 138a:0097
This device is supported

step 1
usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 4d f9 48 9b 41 58  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 02 
Expected at char 012
Sensor not initialized, init byte is 0x2 (expected 0x02)
step 2
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 f9 4b 4a f6  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 f9 4b 4a f6  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
step 4
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6 
01e0 31 28 dc 82 45 
usb read:
0000 be 04 
Expected at char 000
0000 be 04 
step 5
usb write:
0000 3e 
usb read:
[... 804 lines cut ...]
0360 00 00 00 00 00 00 00 00  00 00 00 00 00 
usb read:
0000 01 04 
Expected len: 298, but got 2
0000 01 04 
step 12
usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 4d f9 48 9b 41 58  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 02 
Expected at char 011
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 4d f9 48 9b 41 58  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 02 
step 13
usb write:
0000 1a 
usb read:
0000 91 04 
Expected at char 000
0000 91 04 
Incomplete reverse engineering
```

F0x06, 2018-02-20:
Revision: 53464cb54eef4b9ddfe1e8337930571007545463
`138a:0097` on `ThinkPad X270 (Type 20HN, 20HM)`

```
Prototype version 15
Found device 138a:0097
This device is supported

step 1
usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 22 40 08 98 41 5d  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 02 
Expected at char 012
Sensor not initialized, init byte is 0x2 (expected 0x02)
step 2
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 9f 74 81 8d  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 9f 74 81 8d  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
step 4
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6 
01e0 31 28 dc 82 45 
usb read:
0000 be 04 
Expected at char 000
0000 be 04 
step 5
usb write:
[... 806 lines cut ...]
0360 00 00 00 00 00 00 00 00  00 00 00 00 00 
usb read:
0000 01 04 
Expected len: 298, but got 2
0000 01 04 
step 12
usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 22 40 08 98 41 5d  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 02 
Expected at char 011
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 22 40 08 98 41 5d  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 02 
step 13
usb write:
0000 1a 
usb read:
0000 91 04 
Expected at char 000
0000 91 04 
Incomplete reverse engineering
```

alexanderad, 2018-02-20:
I also have ThinkPad X1 5th generation (138a:0097), prototype and LED test works (against in DB fingerprint), master at 53464cb54eef4b9ddfe1e8337930571007545463
https://gist.github.com/alexanderad/f5f9cb25ff8eae46c4adf188afbc3deb