nmikhailov/Validity90 #24: Cannot build.

yunc5, 2018-01-11:
I'm using X1 Carbon Gen4, with a 138a:0090 sensor, running Ubuntu 16.04.

I'm pretty sure this is a newbie issue and there's something wrong with my system settings, since I'm new to linux. 

When I tried to use './configure', it gives me 

> bash: ./configure: no such file or directory

I searched on google and it said I should use autoconf first. And then I received:

> configure.ac:2: error: possibly undefined macro: AM_INIT_AUTOMAKE
>         If this token and others are legitimate, please use m4_pattern_allow.
>         See the Autoconf documentation.
>configure.ac:12: error: possibly undefined macro: AC_PROG_LIBTOOL
>configure.ac:14: error: possibly undefined macro: AM_PROG_CC_C_0
>configure.ac:166: error: possibly undefined macro: AM_CONDITIONAL

I'm not sure what the last two lines mean but I do know that I have the latest libtool installed. (2.4.6-0.1)

And I'm not actually sure if this program is ready for use or not? Seems it runs as prototype for 138a:0090, so I guessed I could build and use it?

Depaulicious, 2018-01-11:
Why are you trying to run `configure`? The driver is not ready, nor there is autotools support. If you want to build the prototype, check the documentation in its directory.

yunc5, 2018-01-11:
Sorry I didn't know. I never used git or linux before.

closed 2018-02-10 by nmikhailov