nmikhailov/Validity90 #27: Gentoo on Thinkpad P51 make libfprint get ‘VFS_USB_TIMEOUT’ undeclared

wminyu, 2018-02-11:
make[1]: Entering directory '/opt/src/Validity90/libfprint'
Making all in libfprint
make[2]: Entering directory '/opt/src/Validity90/libfprint/libfprint'
  CC       drivers/libfprint_la-vfs0090.lo
drivers/vfs0090.c: In function ‘async_write’:
drivers/vfs0090.c:82:46: error: ‘VFS_USB_TIMEOUT’ undeclared (first use in this function)
                   async_write_callback, ssm, VFS_USB_TIMEOUT);
                                              ^~~~~~~~~~~~~~~
drivers/vfs0090.c:82:46: note: each undeclared identifier is reported only once for each function it appears in
drivers/vfs0090.c: In function ‘async_read’:
drivers/vfs0090.c:127:15: error: ‘EP3_IN’ undeclared (first use in this function)
     if (ep == EP3_IN)
               ^~~~~~
drivers/vfs0090.c:130:28: error: ‘VFS_USB_TIMEOUT’ undeclared (first use in this function)
                            VFS_USB_TIMEOUT);
                            ^~~~~~~~~~~~~~~
drivers/vfs0090.c: In function ‘send_control_packet_ssm’:
drivers/vfs0090.c:148:30: error: ‘struct vfs_dev_t’ has no member named ‘control_packet’
         async_write(ssm, vdev->control_packet, VFS_CONTROL_PACKET_SIZE);
                              ^~
drivers/vfs0090.c:148:48: error: ‘VFS_CONTROL_PACKET_SIZE’ undeclared (first use in this function)
         async_write(ssm, vdev->control_packet, VFS_CONTROL_PACKET_SIZE);
                                                ^~~~~~~~~~~~~~~~~~~~~~~
drivers/vfs0090.c:157:17: error: ‘struct vfs_dev_t’ has no member named ‘control_packet’
         if (vdev->control_packet == next_receive_1) {
                 ^~
drivers/vfs0090.c:157:37: error: ‘next_receive_1’ undeclared (first use in this function)
         if (vdev->control_packet == next_receive_1) {
                                     ^~~~~~~~~~~~~~
drivers/vfs0090.c:158:17: error: ‘struct vfs_dev_t’ has no member named ‘control_packet’
             vdev->control_packet = next_receive_2;
                 ^~
drivers/vfs0090.c:158:36: error: ‘next_receive_2’ undeclared (first use in this function)
             vdev->control_packet = next_receive_2;
                                    ^~~~~~~~~~~~~~
drivers/vfs0090.c:163:26: error: ‘commit_out’ undeclared (first use in this function)
         async_write(ssm, commit_out, sizeof(commit_out));
                          ^~~~~~~~~~
drivers/vfs0090.c:167:34: error: ‘VFS_COMMIT_RESPONSE_SIZE’ undeclared (first use in this function)
         commit_result = g_malloc(VFS_COMMIT_RESPONSE_SIZE);
                                  ^~~~~~~~~~~~~~~~~~~~~~~~
drivers/vfs0090.c:175:32: error: ‘struct vfs_dev_t’ has no member named ‘interrupt’
         async_read(ssm, 3, vdev->interrupt, VFS_INTERRUPT_SIZE);
                                ^~
drivers/vfs0090.c:175:45: error: ‘VFS_INTERRUPT_SIZE’ undeclared (first use in this function)
         async_read(ssm, 3, vdev->interrupt, VFS_INTERRUPT_SIZE);
                                             ^~~~~~~~~~~~~~~~~~
drivers/vfs0090.c:181:18: error: ‘struct vfs_dev_t’ has no member named ‘interrupt’
             (vdev->interrupt, empty_interrupt, VFS_INTERRUPT_SIZE)) {
                  ^~
drivers/vfs0090.c:181:31: error: ‘empty_interrupt’ undeclared (first use in this function)
             (vdev->interrupt, empty_interrupt, VFS_INTERRUPT_SIZE)) {
                               ^~~~~~~~~~~~~~~
drivers/vfs0090.c:187:9: error: implicit declaration of function ‘async_abort’ [-Werror=implicit-function-declaration]
         async_abort(ssm, 3);
         ^~~~~~~~~~~
drivers/vfs0090.c:192:17: error: ‘struct vfs_dev_t’ has no member named ‘control_packet’
         if (vdev->control_packet != turn_on)
                 ^~
drivers/vfs0090.c:192:37: error: ‘turn_on’ undeclared (first use in this function)
         if (vdev->control_packet != turn_on)
                                     ^~~~~~~
drivers/vfs0090.c:193:13: error: implicit declaration of function ‘clear_ep2’ [-Werror=implicit-function-declaration]
             clear_ep2(ssm);
             ^~~~~~~~~
drivers/vfs0090.c: In function ‘activate_ssm’:
drivers/vfs0090.c:225:9: error: implicit declaration of function ‘send’ [-Werror=implicit-function-declaration]
         send(blob_init_s_1);
         ^~~~
drivers/vfs0090.c:221:23: warning: unused variable ‘vdev’ [-Wunused-variable]
     struct vfs_dev_t *vdev = idev->priv;
                       ^~~~
drivers/vfs0090.c:220:34: warning: unused variable ‘udev’ [-Wunused-variable]
     struct libusb_device_handle *udev = idev->udev;
                                  ^~~~
At top level:
drivers/vfs0090.c:206:13: warning: ‘send_packet’ defined but not used [-Wunused-function]
 static void send_packet(struct fpi_ssm *ssm)
             ^~~~~~~~~~~
In file included from drivers/vfs0090.c:28:0:
drivers/vfs0090.h:130:22: warning: ‘blob_init_s_7’ defined but not used [-Wunused-variable]
 static unsigned char blob_init_s_7[] = {
                      ^~~~~~~~~~~~~
drivers/vfs0090.h:126:22: warning: ‘blob_init_s_6’ defined but not used [-Wunused-variable]
 static unsigned char blob_init_s_6[] = {
                      ^~~~~~~~~~~~~
drivers/vfs0090.h:122:22: warning: ‘blob_init_s_5’ defined but not used [-Wunused-variable]
 static unsigned char blob_init_s_5[] = {
                      ^~~~~~~~~~~~~
drivers/vfs0090.h:58:22: warning: ‘blob_init_s_4’ defined but not used [-Wunused-variable]
 static unsigned char blob_init_s_4[] = {
                      ^~~~~~~~~~~~~
drivers/vfs0090.h:54:22: warning: ‘blob_init_s_3’ defined but not used [-Wunused-variable]
 static unsigned char blob_init_s_3[] = {
                      ^~~~~~~~~~~~~
drivers/vfs0090.h:50:22: warning: ‘blob_init_s_2’ defined but not used [-Wunused-variable]
 static unsigned char blob_init_s_2[] = {
                      ^~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [Makefile:1151: drivers/libfprint_la-vfs0090.lo] Error 1
make[2]: Leaving directory '/opt/src/Validity90/libfprint/libfprint'
make[1]: *** [Makefile:477: all-recursive] Error 1
make[1]: Leaving directory '/opt/src/Validity90/libfprint'
make: *** [Makefile:386: all] Error 2

nmikhailov, 2018-02-12:
libfprint driver is not ready, you can test prototype in the mean time.

closed 2018-02-12 by nmikhailov

jontis, 2018-09-21:
nmikhailov has not been active the last 3 months. No idea why.