nmikhailov/Validity90 #28: Prototype make not working - gcrypt.h not found

dblitt99, 2018-02-14:
Hi, I am trying to test the prototype of the driver on my Dell XPS 13 9360. I cloned the repo and ran `make` in the prototype directory. This is the output:
```
cc -c -Wall -g -DG_LOG_DOMAIN=\"Validity90\" -I/usr/include/nss3 -I/usr/include/nspr4 -I/usr/include/libusb-1.0 -I/usr/include/libpng16 -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  validity90/utils.c -o validity90/utils.o
validity90/utils.c:23:10: fatal error: gcrypt.h: No such file or directory
 #include <gcrypt.h>
          ^~~~~~~~~~
compilation terminated.
make: *** [Makefile:31: validity90/utils.o] Error 1

padok, 2018-02-14:
Did you check if you have installed the developer version of the library?
```sudo dnf install libgcrypt-devel```

dblitt99, 2018-02-14:
@padok well that does it! I guess I should have checked that first.

closed 2018-02-14 by dblitt99