nmikhailov/Validity90 #31: Prototype output (138a:0097 Validity Sensors, Inc.)

MatthiasLohr, 2018-02-23:
Hello,

the current prototype tool works out of the box on my Carbon X1 (5th gen). I ran fingerprint read an LED test. Please let me know if i can provide further information.

Here is the prototype output:
```
Prototype version 15
Found device 138a:0097
This device is supported

step 1
usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 05 51 08 de 45 5f  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 07 
Expected at char 012
step 2
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 b5 3d 69 48  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
step 3
usb write:
0000 43 02 
usb read:
0000 00 00 01 00 01 00 08 00  5e 78 2d 59 01 00 34 46 
0010 02 00 07 00 30 3e 00 00  01 00 84 08 01 00 07 00 
0020 00 04 00 00 02 00 84 28  03 00 12 00 e0 10 00 00 
0030 02 00 76 36 01 00 0c 00  10 0a 00 00 01 00 86 47 
0040 00 00 01 00 50 5a 00 00  02 00 23 77 00 00 01 00 
0050 80 2f 00 00 02 00 66 37  01 00 0c 00 f0 22 02 00 
0060 01 00 25 56 00 00 01 00  70 04 00 00 
Expected len: 84, but got 108
step 4
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
[... 1132 lines cut ...]
0010 93 17 2d 54 ef 17 35 49  1d 4a da c1 ed 88 b3 c7 
0020 c0 8c 9d 67 5c 03 2a db  f0 11 eb 61 a0 31 0a 77 
0030 9d 6d ca e7 7f b4 06 50  7a 06 a5 78 7a cf 95 a5 
0040 43 f4 ee 16 f4 a1 3b 92  a7 bb a1 f1 c5 b4 43 28 
0050 02 0a d8 cd bb 4d 3a 28  0e 81 09 c4 5d 0d c8 26 
0060 79 93 7b 5b 23 99 a1 5a  97 f7 0d b4 25 6a 6d 0c 
0070 80 e8 73 17 51 40 0b 12  94 69 d2 56 fb 66 97 2b 
0080 64 10 49 65 6b 0f 78 43  a6 f7 c7 d6 34 9f 6d 57 
0090 32 4b 26 d3 af 7a f6 51  bb bd 9d 4e 6e a6 b0 d4 
00a0 67 70 50 ba e3 d6 8c 64  51 13 4c 18 24 ad 38 26 
00b0 4d 19 21 c7 99 
usb read:
0000 17 03 03 00 40 43 b1 8a  b8 2c cd 27 a0 e7 80 9b 
0010 10 d6 de ab 72 33 f9 75  d4 95 82 6c 41 c3 f3 9e 
0020 d0 5b 6c e2 67 6b be dc  c4 d0 a4 bb f7 1e 7f c8 
0030 0e a4 d9 c8 e9 24 85 05  91 65 77 c7 b6 f7 ac 60 
0040 ea 30 06 a8 2a 
READ:
0000 00 00 

1 - Scan fingerprint
2 - Test leds
0 - Exit
0
```

jkrauska, 2018-02-28:
Same/similar results.  Let me know if you want details.

        Manufacturer: LENOVO
        Product Name: 20HRS15700
        Version: ThinkPad X1 Carbon 5th

zsedem, 2018-04-12:
Works on Lenovo Thinkpad T470 :+1:

rdpascua, 2018-05-03:
Same issue here

Manufacturer: Lenovo
Version: Thinkpad T470p