nmikhailov/Validity90 #33: Offer Help with a T470 20HE

ficofer, 2018-03-11:
@nmikhailov 

Hello guys, how can I help debug and support the fingerprint reader integration for the:

```Bus 001 Device 005: ID 138a:0097 Validity Sensors, Inc. ```

Running Ubuntu 16.04.
I am not sure if this repo is a complete implementation in its beta phases or just a deploy of the code?

ficofer, 2018-03-11:
```Bus 001 Device 005: ID 138a:0097 Validity Sensors, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass        16 
  bDeviceProtocol       255 
  bMaxPacketSize0         8
  idVendor           0x138a Validity Sensors, Inc.
  idProduct          0x0097 
  bcdDevice            1.64
  iManufacturer           0 
  iProduct                0 
  iSerial                 1 d617888954d9
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           53
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           5
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval              10
Device Status:     0x0000
  (Bus Powered)
```

nmikhailov, 2018-03-12:
The actual status is available in the README. Right now you can check if prototype works or not.

erdenezul, 2018-03-13:
hope it helps
<pre>
Prototype version 15
Found device 138a:0097
This device is supported

step 1
usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 04 04 c8 91 45 a2  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 07 
Expected at char 012
step 2
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 8e a4 af dd  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
step 3
usb write:
0000 43 02 
usb read:
0000 00 00 01 00 01 00 08 00  5e 78 2d 59 01 00 34 46 
0010 02 00 07 00 30 3e 00 00  01 00 84 08 01 00 07 00 
0020 00 04 00 00 02 00 84 28  03 00 12 00 e0 10 00 00 
0030 02 00 76 36 01 00 0c 00  10 0a 00 00 01 00 86 47 
0040 00 00 01 00 50 5a 00 00  02 00 23 77 00 00 01 00 
0050 80 2f 00 00 02 00 66 37  01 00 0c 00 f0 22 02 00 
0060 01 00 25 56 00 00 01 00  70 04 00 00 
Expected len: 84, but got 108
step 4
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6 
01e0 31 28 dc 82 45 
usb read:
0000 be 04 
[... 1034 lines cut ...]
0000 17 03 03 00 b0 4b 77 62  ff a9 03 c1 1e 6f d8 35 
0010 93 17 2d 54 ef 8a 8a 65  b1 dd 6f 2b 81 f4 d3 d1 
0020 cf fa 99 6b 96 f6 5a 4d  07 97 59 6d 26 33 2b 3c 
0030 48 7e 80 72 02 8a 62 21  f3 14 85 4a 11 12 33 10 
0040 6b 41 4a b0 7d 1b 36 42  13 d3 09 ac 92 b8 78 b5 
0050 65 52 29 56 2a 5c 77 ad  be 78 62 c0 71 c0 68 02 
0060 0b 85 bc 14 89 96 33 67  ed 8c dc f3 bf 73 a3 c5 
0070 2f bf c6 57 9c b4 6e c2  0c fc 44 9c 76 63 f5 ec 
0080 95 ea 2f 51 13 a5 22 53  a3 36 e2 5a c4 7d e2 c7 
0090 b5 9d ac 2f 0b 2b 9b df  f5 20 c7 73 9a 98 52 98 
00a0 2d 1c 4c 21 5d 27 75 0f  b7 31 93 02 69 a0 a8 fd 
00b0 7f 9f b9 5b b8 
usb read:
0000 17 03 03 00 40 9f f8 bb  b7 22 11 0e cb 39 17 b3 
0010 35 c4 14 79 c8 0f ca 7a  57 b6 a9 6b 65 7a 94 91 
0020 83 9d 70 fa 02 7d d5 5a  b6 78 39 49 3b ba cd 9d 
0030 51 25 ae 54 d1 9f 23 09  d8 b9 23 c7 99 32 4a bc 
0040 44 e6 d7 b8 ac 

Fingerprint UNKNOWN!

1 - Scan fingerprint
2 - Test leds
0 - Exit
</pre>

zsedem, 2018-04-12:
here is another output https://gist.github.com/zsedem/a02c365c3848ee97fef24611a6f9a1b0

closed 2020-05-03 by ficofer