nmikhailov/Validity90 #34: Add support for 06cb:009a

frederictobiasc, 2018-03-13:
Hi, Lenovos T480 uses another unsupported fingerprint reader. I changed the main.c of the prototype in order to execute the tasks. Initialisation and "Test leds" works fine, but "Scan fingerprint" seems to be not working.
 [Prototype log](https://github.com/frederictobiasc/Validity90/blob/bfdce77b7be59b06f81e5c453c060d1fcf191e49/prototype/logs/log9a)

nmikhailov, 2018-03-13:
Try replacing https://github.com/nmikhailov/Validity90/blob/master/prototype/main.c#L922 with your last code

nmikhailov, 2018-03-13:
I have pushed small update, can you check it?

nmikhailov, 2018-03-13:
I have pushed some test code in 9a branch. Could you check it out please?

frederictobiasc, 2018-03-13:
Hey, I applied the fix as mentioned before, I'll check out 9a now. You can find the new logfile at https://github.com/frederictobiasc/Validity90/blob/master/prototype/logs/log9a
It is crashing with a segmentation fault at line 993.

frederictobiasc, 2018-03-13:
I checked out the code and pushed the new logfile. The last output is:
`Failed 'libusb_bulk_transfer(dev, 0x81, data, len, out_len, 10000)': -1 - LIBUSB_ERROR_IO`

nmikhailov, 2018-03-14:
Updated again, please check

frederictobiasc, 2018-03-14:
Hi,
now it is working without any errors.
[log](https://github.com/frederictobiasc/Validity90/blob/5472b9cedcf7a5ffdb64671ea5e2dac886713a82/prototype/logs/9a)
The problem is, that this process happens without me putting any finger on the reader. There is not waiting at "Waiting for finger"

`Awaiting fingerprint:
interrupt:
0000 00 00 00 00 00 
Waiting for finger...
interrupt:
0000 02 00 40 10 00 
Finger is on the sensor...
interrupt:
0000 03 40 01 00 00 
Scan in progress...
interrupt:
0000 03 42 04 00 40 
Scan succeeded! (v97)`

Even if I put my finger on the reader, the output is "Fingerprint unknown".
Would it help if I go on with dissecting the traffic between windows and the reader during the scanning process?

nmikhailov, 2018-03-14:
Do you have any fingers enrolled in windows?
Dumps won't help much I am afraid. All the traffic is encrypted.

frederictobiasc, 2018-03-14:
Yes, fingerprints are enrolled in Windows and working. I assumed that I could decrypt the traffic with the [dissector.lua](https://github.com/nmikhailov/Validity90/blob/7ed2c1043e41ca5438cf54f11bc5ed8ffc640b06/dissector.lua).

vikramambrose, 2018-03-15:
Just FYI this affects many new Thinkpad models. 

T480/T480s, X1 Carbon 6th gen, X1 Yoga 3rd gen, X280

nmikhailov, 2018-03-26:
@frederictobiasc Ok, here is what we can do:
1. You should get Process explorer from here - https://docs.microsoft.com/en-us/sysinternals/downloads/process-explorer
2. Detach usb reader from VM.
3. Attach it and search for newly created WUDFHost.exe process in process explorer. Be careful as it might be several of them.
4. Start wireshark sniffing
5. Do fingerprint scan
6. Stop sniffing, save logs
7. In process explorer right click on WUDFHost.exe -> Create Dump -> Full Dump.
Upload traffic and dumps somewhere.

frederictobiasc, 2018-03-26:
@nmikhailov Okay, I will provide you with the requested dumps ASAP. :)

erpalma, 2018-04-03:
It's interesting that after running the prototype, all the stored fingerprints are lost. I don't use Windows but I use the pre-boot EFI scan. Also on Windows it seems that if you have unlocked the fingerprint in EFI, then you don't need to scan again. It seems that there exists a way to just "ask" the reader if it has already been unlocked, right?

nmikhailov, 2018-04-03:
Well, it's actually quite strange that you have lost enrolled fingerprints,
it shouldn't happen unless  you get "incomplete reverse engineering"
message.

I haven't investigated " previous scan info mechanism" you are talking
about, although it is very much possible it exist, I can even think of a
way how it could have been done API wise.

Can you write some more about bios fingerprint related settings? What is
possible and what is not. I have a feeling bios handling has progressed
sine my time.

erpalma, 2018-04-06:
In bios you can just enable fingerprint authentication. You can set only if the bios will also ask for the supervisor pwd or just the fingerprint. I believe that, once enabled, the bios will ask you for the fingerprint if at least one was enrolled. At Windows boot I see the reader white led turning on for half a second and then the desktop appears. If I skip the fingerprint reading during bios (using pwd instead) then Windows ask me the finger. Right now I don't have a W10 VM but I can create one. Do you know if I can sniff the USB bus on Linux while a device is attached to the VM?

nmikhailov, 2018-04-06:
> Do you know if I can sniff the USB bus on Linux while a device is
attached to the VM?

Yes, that it is what I am doing.

mvdnes, 2018-04-06:
I have not seen it in the discussion here, so I would like to mention that this reader is advertised as "Match-on-chip touch fingerprint reader".

The prototype is able to blink the LED's. Furthermore, in my case the fingerprint in the EFI setup menu still worked.

Unfortunately I cannot get the fingerprint reader working in a VM. I get this error: "This device is not working properly because Windows cannot load the drivers required for this device. (Code 31)". (This was after installing the official drivers, before it was not recognized at all)

nmikhailov, 2018-04-06:
@mvdnes Can you link driver download url? I would think one of the reasons why this might have happened is that they pushed for mandatory SGX.

mvdnes, 2018-04-06:
You can download it from this [lenovo download page]. Here is a [direct link].

[lenovo download page]: https://pcsupport.lenovo.com/nl/en/products/LAPTOPS-AND-NETBOOKS/THINKPAD-T-SERIES-LAPTOPS/THINKPAD-T480-TYPE-20L5-20L6/20L5/downloads/DS502265
[direct link]: https://download.lenovo.com/pccbbs/mobiles/n20gf06w.exe

nmikhailov, 2018-04-06:
@mvdnes It still includes non-sgx version, so I don't think that it is a problem. Have you tried rebooting? Are you sure you have passthough correct usb device to VM?

mvdnes, 2018-04-06:
Using another virtualization platform (VirtualBox istead of qemu/kvm) worked.

I have a [wireshark dump] and a [WUDFHost dump] for you.

[wireshark dump]: https://www.mvdnes.nl/files/validity90/scan1.pcapng
[WUDFHost dump]: https://www.mvdnes.nl/files/validity90/scan1_WUDFHost.dmp.xz

mvdnes, 2018-05-04:
I have used your cdb script to extract the AES keys myself. The dissector this repository provides works on it!

Here is the [log archive].
It contains a wireshark dump, debug trace and extracted AES keys.

The log was taken while unlocking Windows.

[log archive]: https://www.mvdnes.nl/files/validity90/log-2018-05-04T21:31:17+02:00.zip

stackpivot, 2018-08-02:
@nmikhailov Owner of T480s here. What would be the best way helping for 06cb:009a to work?

felixonmars, 2018-08-09:
After changing the id matching part in main.c, I got the following error:
```
Failed 'EC_KEY_check_key(key) - 1': -1 - LIBUSB_ERROR_IO
```

Full output: https://paste.xinu.at/0bMCm/

spielkind, 2018-08-10:
@mrdanielps 
I own a X280: Bus 001 Device 007: ID 06cb:009a Synaptics, Inc. 
With your 9a branch I get:
Prototype version 15
No devices found

With your 9d branch I get: 
Fingerprint UNKNOWN!

He do not even wait until I put my finger on the scanner.

mrdanielps, 2018-08-10:
So if instead of repeating the whole scan procedure, I send a `StgWindsor` packet (`4b 00 00...`, or `packet4` in the code) before waiting for the match result interrupt, it seems to work properly (at least on 9d). [This commit](https://github.com/mrdanielps/Validity90/commit/c726bb944d9c2a60dcc6a60465985b9323de74b3) is the only change in master to get it to work.

@spielkind 
Weird. It's actually detecting a finger and returning a mismatch. From one of the previous logs, without these changes, the 9a behaved similar to my model.

spielkind, 2018-08-10:
Dunno, I've resettet all fingerprint data, then I get: 'Incomplete reverse engineering' ... after enrolling a new finger at my win10 vm (virtualbox) I get "Fingerprint UNKNOWN' again, without waiting for touching the sensor.

tedsluis, 2018-08-11:
note: Thinkpad T580 is using this fingerprint reader as well:

(Windows)
Synaptics WBDI, class=0xFF, subclass=0x10, vendor=0x06CB, product=0x009A

(Linux)
lsusb  -d 06cb:009a
Bus 001 Device 006: ID 06cb:009a Synaptics, Inc.

ramonmaruko, 2018-11-04:
Hello! What data do you still need to support 009a?

Zlogene, 2018-11-09:
@ibrahima, they clearly state that Ubuntu flawors may not work as expected without additional chdnges from a user, and as Ubuntu uses libfprint (as many others), which has no the driver yet, their certification means nearly nothing

cnmicha, 2018-11-09:
@ibrahima Lenovo L380 is also affected and not on the list.

cgarlati, 2018-11-23:
I have an X1 6th gen. Let me know if I can help in any way.

mgiacopu, 2018-11-23:
I have a ThinkPad T580 which also uses this kind fingerprint reader. Let me know if I could be of any help, I'd really like to get it working also on Linux!

noizo, 2018-12-03:
Hi.
I have this reader on Lenovo P52
```
lsusb -d 06cb:009a -v

Bus 001 Device 003: ID 06cb:009a Synaptics, Inc. 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass        16 
  bDeviceProtocol       255 
  bMaxPacketSize0         8
  idVendor           0x06cb Synaptics, Inc.
  idProduct          0x009a 
  bcdDevice            1.64
  iManufacturer           0 
  iProduct                0 
  iSerial                 1 
  bNumConfigurations      1
```

How can i help with test?

AndreMilani, 2018-12-05:
Hi, 
hoping this can help I share some info about my reader on ThinkPad T480.
Let me know if I can help in any way.
```
Bus 001 Device 005: ID 06cb:009a Synaptics, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass        16 
  bDeviceProtocol       255 
  bMaxPacketSize0         8
  idVendor           0x06cb Synaptics, Inc.
  idProduct          0x009a 
  bcdDevice            1.64
  iManufacturer           0 
  iProduct                0 
  iSerial                 1 40069486a5e0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           53
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           5
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval              10
Device Status:     0x0000
  (Bus Powered)
```

AxeR21, 2018-12-10:
I have a ThinkPad T480. I'm studying computer science at the university so, if I can help in some way, let me know and I'll try to make my best.
Thanks for your hard work

federeghe, 2018-12-10:
This affects also Thinkpad L480.

Let us know how we can help! :)

RaghavRao, 2018-12-10:
Also here with X1 Yoga 3rd Gen.

Is this project still active?

ix5, 2018-12-10:
Jesus people stop spamming this!
Either take a look at the code or don't, some of us have email notifications set up for this and do not want to wade through tens of useless "me too" messages.

anp369, 2019-01-10:
@mrdanielps I don't know if you're still interested but I got your prototype in the 9d branch (commit  d7789c8 ) working with 138a:009d (Yoga X380) I enrolled the fingerprints in Windows before and the prototype was able to recognize my fingerprint again. leds work too. Let me know if you need any logs from this sensor

scaleoutsean, 2019-01-21:
> I don't know if you're still interested but I got your prototype in the 9d branch (commit [d7789c8](https://github.com/nmikhailov/Validity90/commit/d7789c8c347811ebbb3af70604869adba18b989c) ) working with 138a:009d (Yoga X380) 

@anp369 thanks for trying. As the other guy is MIA, I tried your patch on T480 and got the same output as with `master` (as of now). On my T480 it's device `Bus 001 Device 004: ID 06cb:009a Synaptics, Inc. `.

```
Prototype version 15
Found device 06cb:009a

step 1
usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 75 34 49 68 58 30  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 02 
Expected at char 012
Sensor not initialized, init byte is 0x2 (expected 0x02)
step 2
usb write:
...
```
There are few other discrepancies ("expected X bug got Y...") It's the same device and same output as with non-patched code, so I won't paste the entire output here as it probably doesn't add much value. But in the case you want to see it, I uploaded it to https://pastebin.com/SuGrT6nS (for 14 days).

pcraciunoiu, 2019-01-21:
Hi all, joining this thread as I'd like to help get this worked out with testing, debug output, etc.

I have a ThinkPad X1 Carbon 6th gen from late last year.

If I `make && make permissions && sudo ./prototype` on master, I get the device:
```
Prototype version 15
Found device 06cb:009a
```

The 2nd option to test leds seems to work with green (white light in my case) and red blinking 3x.

1st option for scan fingerprint doesn't wait for the scan, or go back to the menu, so I am guessing it doesn't work. Happy to provide more output if it'd be helpful.

Curiously, if I check out the `9a` branch, my device isn't found. Which seems wrong since clearly it's found right above. Listed on `lsusb`:

```
Bus 001 Device 005: ID 06cb:009a Synaptics, Inc. 
```

pcraciunoiu, 2019-01-21:
Have not tried a Windows VM--not sure what the value of that is since I want it to work in Ubuntu? When I got the laptop I had set up the fingerprint reader on Windows 10 (host) and it worked fine.

felixonmars, 2019-01-21:
You can capture the USB traffic if you use virtualbox. I am not expert though, and no progress so far to improve the prototype.

mid-kid, 2019-03-06:
Sorry for bumping this thread, but I've followed the instructions in https://github.com/nmikhailov/Validity90/issues/34#issuecomment-376230153, and here's the results:
https://transfer.sh/LnWdg/WUDFHost.dmp
https://transfer.sh/J9y9M/fprint.pcapng
Hopefully this helps a bit.

knz, 2019-05-10:
Using x380 and  138a:009d reader: the code linked from this comment https://github.com/nmikhailov/Validity90/issues/34#issuecomment-411724539 works fine!

@mrdanielps @nmikhailov any chance we could integrate this further in the driver?

mrdanielps, 2019-05-11:
> Using x380 and 138a:009d reader: the code linked from this comment [#34 (comment)](https://github.com/nmikhailov/Validity90/issues/34#issuecomment-411724539) works fine!
> 
> @mrdanielps @nmikhailov any chance we could integrate this further in the driver?

If you're asking about libfprint, I guess I never linked to my own fork over [here](https://github.com/mrdanielps/libfprint). Ignore the PPA in the README, it must be built from source.
I should mention it's only been tested on a 138a:009d reader with the fingerprints enrolled in a VM.

vitaliis, 2019-05-11:
> I should mention it's only been tested on a 138a:009d reader with the fingerprints enrolled in a VM.

It provides at least some hope, that people could start using fingerprint sensor in their machines.

Moving this forward to libfprint is highly appreciated!

JohnAZoidberg, 2019-06-19:
This is what it does on master(8b80cfeae15487d5e13ff5ff3ab4f282eac765be) for me, without touching the reader at all:

<details>

  ```
  $ ./prototype
Prototype version 15
Found device 06cb:009a

step 1
usb write:
0000 01
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01
0010 00 00 dc 3f 89 68 47 a4  00 23 00 00 00 00 01 00
0020 00 00 02 d1 00 07
Expected at char 012
step 2
usb write:
0000 19
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00
0010 00 00 00 00 75 96 48 d7  00 00 00 00 00 00 00 00
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
0040 00 00 00 00
step 3
usb write:
0000 43 02
usb read:
0000 00 00 01 00 02 00 08 00  8a 7d d9 5b 01 00 34 46
0010 02 00 07 00 e0 42 00 00  01 00 84 08 01 00 07 00
0020 00 04 00 00 02 00 84 28  03 00 12 00 40 10 00 00
0030 02 00 76 36 01 00 0c 00  10 0a 00 00 01 00 86 47
0040 00 00 01 00 50 5a 00 00  02 00 23 77 00 00 01 00
0050 80 2f 00 00 02 00 00 37  00 00 01 00 b0 08 02 00
0060 01 00 25 56 00 00 01 00  60 04 00 00
Expected len: 84, but got 108
step 4
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6
[... 932 lines cut ...]
interrupt:
0000 02 00 40 10 00
Finger is on the sensor...
interrupt:
0000 03 40 01 00 00
Scan in progress...
interrupt:
0000 03 41 03 00 40
Fingerprint scan completed...
^C
  ```

</details>

then it's stuck there and 

and this is on 9a (59a99b2295753e1a63edd343658f49ff947c403b)

  ```
$ ./prototype
Prototype version 15
No devices found
  ```

Of course my device is 06cb:009a and I'm running Linux 5.1.8

point-source, 2019-06-20:
I have a similar result as @JohnAZoidberg but different messages

master branch gives me this:

<details>
Prototype version 15
Found device 06cb:009a

step 1
usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 15 bc 4b 68 4c 64  00 23 00 00 00 00 01 00 
0020 00 00 02 d1 00 02 
Expected at char 012
Sensor not initialized, init byte is 0x2 (expected 0x02)
step 2
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 49 28 e0 7d  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 49 28 e0 7d  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
step 4
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6 
01e0 31 28 dc 82 45 
usb read:
0000 be 04 
Expected at char 000
0000 be 04 
step 5
usb write:
[... 817 lines cut ...]
0010 00 00 15 bc 4b 68 4c 64  00 23 00 00 00 00 01 00 
0020 00 00 02 d1 00 02 
Expected at char 011
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 15 bc 4b 68 4c 64  00 23 00 00 00 00 01 00 
0020 00 00 02 d1 00 02 
step 13
usb write:
0000 1a 
usb read:
0000 91 04 
Expected at char 000
0000 91 04 
Incomplete reverse engineering
</details>

branch 9a reports:

```
$ ./prototype
Prototype version 15
No devices found
```

I did not touch the reader during either of these tests

wertyy102, 2019-06-25:
As I just stumbled across this today, I haven't done any testing yet.  But the E580 Thinkpad also has the 06cb:00a2 Fingerprint reader.  I'll see how it behaves.  Debian Jessie and Linux Mint are both at my disposal on two separate laptops, same model and specs.  Will update today with at least the Debian results.

wertyy102, 2019-06-25:
Update:  
after setting up all dependencies to compile, and doing so successfully, 

`
Prototype version 15
No devices found
`
But the output of lsusb is:
`
Bus 001 Device 008: ID 06cb:00a2 Synaptics, Inc.
`

wertyy102, 2019-06-25:
Ignore my musings, I have been posting in the wrong thread...

TheAifam5, 2019-07-23:
I noticed that the XMG laptops using that thing.

picaschaf, 2019-07-25:
I got a X1 Carbon 6th gen and really want to get the last bit of hw working. Is there anything that could support the efforts in making a Linux driver possible? Logs, Dev, Money? Anything?

pcraciunoiu, 2019-07-25:
Same here, I've had this laptop 7 months. Would be great to use the fingerprint reader in the next year or two, after that it'll be too late :) Willing to help with QA/logs/donation...

codebam, 2019-07-25:
Same here, X1 6th gen here if anyone needs anything

dirtyhawk, 2019-07-26:
II'm here with a Lenovo T480s, and also willing to help. What can we do?

tgabi333, 2019-07-26:
Im also herw to help with an x1

ros86, 2019-07-26:
Same here with T480

[74 more replies]