nmikhailov/Validity90 #36: prototype hangs after "Finger is on the sensor..." - T470 - 138a:0097

JonasLoos, 2018-03-27:
The prototype seems to run smoothly until nothing happens after the message "Finger is on the sensor...". The LED test works.

* ./prototype 
* 1 - Scan fingerprint 
* wait for "Waiting for finger..." 
* put finger on reader -> "Finger is on sensor..."
* nothing happens (LED is out, no console output for >1min)

output: 
master: https://gist.github.com/JonasLoos/14e79865ea26bfa880bf9b735f90e8c7
9a: https://gist.github.com/JonasLoos/0fbdab33971c9337cf2b5a1c04c774a9
(only hexcode differs)

Is this a bug? If not, how can I get it to work?

nmikhailov, 2018-03-27:
Not sure why that happens. Does it work properly on windows?

You can try replacing https://github.com/nmikhailov/Validity90/blob/master/prototype/main.c#L932
scan_matrix1 wtih v97_scan_matrix2

PS: 9a is not for 138a:0097.

JonasLoos, 2018-03-27:
I don't have Windows installed on hard drive.

After replacing scan_matrix1 wtih v97_scan_matrix2 (and running make) the output didn't change (except for some hexcode), it still didn't work. 
But after an unintended restart and some unsuccessful tries to get it to work with Win 10 in a VM, the prototype now works on Linux (Solus) like intended ("Fingerprint UNKNOWN!"). 
Don't know what caused it but thanks.

nmikhailov, 2018-03-28:
Strange. Probably it wasn't fully/completly initialized in the first place.

closed 2018-03-28 by nmikhailov