nmikhailov/Validity90 #38: Prototyp output for 138a:0097 on Lenovo T470

Isengo1989, 2018-06-09:
Are they still needed?

`Prototype version 15
Found device 138a:0097

step 1
usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 32 50 88 98 45 a1  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 07 
Expected at char 012
step 2
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 db c4 8e df  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
step 3
usb write:
0000 43 02 
usb read:
0000 00 00 01 00 01 00 08 00  dd a6 4e 57 01 00 34 46 
0010 02 00 07 00 60 39 00 00  01 00 84 08 01 00 07 00 
0020 00 04 00 00 02 00 84 28  03 00 12 00 e0 10 00 00 
0030 02 00 76 36 01 00 0c 00  10 0a 00 00 01 00 86 47 
0040 00 00 01 00 50 5a 00 00  02 00 23 77 00 00 01 00 
0050 80 2f 00 00 02 00 66 37  01 00 0c 00 f0 22 02 00 
0060 01 00 25 56 00 00 01 00  70 04 00 00 
Expected len: 84, but got 108
step 4
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6 
01e0 31 28 dc 82 45 
usb read:
0000 be 04 
Expected at char 000
[... 347 lines cut ...]
0100 bf c5 9b 3c 67 d9 45 06  38 da 92 be 65 bf 81 8c 
0110 aa 7e 0f 00 00 48 30 46  02 21 00 e2 3e 59 5d 9f 
0120 6c 3c 70 59 cf 93 e1 35  6b 02 80 c6 06 4e a4 15 
0130 1d 05 84 3e 9d 16 78 05  a6 25 da 02 21 00 b7 18 
0140 26 86 d5 17 8e e8 1d 4f  db 82 a3 7b b1 d1 64 2f 
0150 1c 4c 47 46 a8 5d df 9d  1a 9c fc d8 dc b0 14 03 
0160 03 00 01 01 16 03 03 00  50 4b 77 62 ff a9 03 c1 
0170 1e 6f d8 35 93 17 2d 54  ef ee 1e 28 4f b6 84 39 
0180 14 7b a5 40 7b b7 82 d0  e6 0a 7c 6e 94 7c e1 e3 
0190 1e c6 35 ae 16 6f ac 98  c4 d4 63 01 f8 20 f1 64 
01a0 e7 dc 2e 08 24 6d 2a 85  09 19 08 6f 98 91 f9 4f 
01b0 fe c7 a3 c0 3d 5d 97 1d  fd 
usb read:
0000 14 03 03 00 01 01 16 03  03 00 50 52 02 a2 68 d0 
0010 69 ca 2e f0 fa 8d e7 dc  ca 5c 98 d8 73 11 ae ec 
0020 56 28 15 a7 53 87 83 fe  55 be 02 0b 09 3c 0d 39 
0030 b9 e2 de 20 66 d3 cd b8  06 76 00 cc ea 80 8e 48 
0040 da 82 95 f5 9b 51 12 fa  d6 92 3c 75 58 f2 5c 3f 
0050 fd 5e 65 38 26 89 10 72  c3 4a 39 
IN: a869feb95e4c41abbea2e48b70785f9ecb009423a3006816364d6ff5537f99da
OUT: 177785faef905cc84abd3fc16b653f545f92fd1009d9592d976db086ca33b1e0

1 - Scan fingerprint
2 - Test leds
0 - Exit`

karambir, 2018-07-02:
Adding more for t470

Prototype version 15
Found device 138a:0097

step 1
usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 87 5f 48 35 45 99  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 07 
Expected at char 012
step 2
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 03 6c a6 39  00 00 00 00 fd bf 85 13 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
step 3
usb write:
0000 43 02 
usb read:
0000 00 00 01 00 01 00 08 00  dd a6 4e 57 01 00 34 46 
0010 02 00 07 00 60 39 00 00  01 00 84 08 01 00 07 00 
0020 00 04 00 00 02 00 84 28  03 00 12 00 e0 10 00 00 
0030 02 00 76 36 01 00 0c 00  10 0a 00 00 01 00 86 47 
0040 00 00 01 00 50 5a 00 00  02 00 23 77 00 00 01 00 
0050 80 2f 00 00 02 00 66 37  01 00 0c 00 f0 22 02 00 
0060 01 00 25 56 00 00 01 00  70 04 00 00 
Expected len: 84, but got 108
step 4
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6 
01e0 31 28 dc 82 45 
usb read:
0000 be 04 
Expected at char 000
[... 347 lines cut ...]
0100 bf c5 9b 3c 67 d9 45 06  38 da 92 be 65 bf 81 8c 
0110 aa 7e 0f 00 00 48 30 46  02 21 00 90 f8 1c 52 52 
0120 aa b8 1d 97 43 a3 c0 11  30 8f 84 73 88 8d da f3 
0130 ab 2d a8 bd 02 09 3b 32  32 bf 3f 02 21 00 ba 6d 
0140 0e 2f e5 97 6a ec 2b 7b  de 0b c0 bd 05 24 e4 cf 
0150 60 db 45 26 29 36 b6 8b  ad a7 9a d9 40 eb 14 03 
0160 03 00 01 01 16 03 03 00  50 4b 77 62 ff a9 03 c1 
0170 1e 6f d8 35 93 17 2d 54  ef 25 0e 5b a5 54 24 ee 
0180 ca 6f b7 18 6f 41 5f 8a  cc 6e c6 e7 e9 8d a9 14 
0190 dd c0 81 02 e3 ac a3 1a  b2 37 dc 2b 23 e7 93 3a 
01a0 ce 81 b3 5d 51 59 f6 d7  02 1e 38 0f 31 d2 4d a2 
01b0 e4 b6 b6 70 78 38 a9 58  13 
usb read:
0000 14 03 03 00 01 01 16 03  03 00 50 66 0e b4 1d 55 
0010 52 15 cf 09 b5 eb 67 fb  66 3a d3 16 ea 3b 01 5f 
0020 c1 d4 2a 8b 6e 7c 6c 94  d1 ea c7 ab be bf 60 b2 
0030 09 47 08 0a 71 ac c1 16  aa 87 a4 72 79 09 b8 d0 
0040 7d 64 a4 87 e1 d4 c5 e3  8f 2c 69 b9 b2 33 63 b1 
0050 e8 b3 c2 44 01 e2 b5 0c  01 1d 2b 
IN: 978be6944f4476cd7bb9ad298c37f144e1d6cf3dc513d2f942c008b6c774de99
OUT: d1ce950849c931c0d754f35818490acafde63b21e95a90446badca73803ff75a

1 - Scan fingerprint
2 - Test leds
0 - Exit

mickare, 2019-01-31:
More T470s on a fresh installed Ubuntu 18.10 ...
But I get a `Incomplete reverse engineering` message at the end.

```
$ lsusb -s 001:006
Bus 001 Device 006: ID 138a:0097 Validity Sensors, Inc.
```

<details><summary>Prototype output...</summary>
<p>

```
Prototype version 15
Found device 138a:0097

step 1
usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 ba 67 88 de 45 5f  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 02 
Expected at char 012
Sensor not initialized, init byte is 0x2 (expected 0x02)
step 2
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 19 73 62 f6  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 19 73 62 f6  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
step 4
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6 
[... 815 lines cut ...]
0000 01 04 
Expected len: 298, but got 2
0000 01 04 
step 12
usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 ba 67 88 de 45 5f  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 02 
Expected at char 011
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 ba 67 88 de 45 5f  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 02 
step 13
usb write:
0000 1a 
usb read:
0000 91 04 
Expected at char 000
0000 91 04 
Incomplete reverse engineering
```
</p>
</details>

closed 2023-10-11 by Isengo1989