nmikhailov/Validity90 #46: Add support for Kensington VeriMark (06cb:0088)

TheCubeMaster019, 2018-09-02:
This device also seems to fit in with this project, as the wireshark dissector recognizes the traffic. I ran it through prototype, and the output is attached to this post.

Thanks a lot!
Prism
[out.txt](https://github.com/nmikhailov/Validity90/files/2343383/out.txt)

OwenPemberton, 2019-02-08:
Would also be interested in getting this working. 
I'm getting "Expert Info (Error/Malformed): Vector length 2 is too large, truncating it to 0"
on the 'elliptic curve points format length' part of the TLS 'client hello' packet.

IncPlusPlus, 2019-04-10:
I've been looking everywhere for a way to use my VeriMark reader to work in Ubuntu. I don't know how to contribute to this process, though. If I can provide any useful info to move this issue along, please let me know and I'd be happy to help.

elmcrest, 2020-11-09:
Hey everyone,

any updates on this?

jontis, 2020-11-09:
#94