nmikhailov/Validity90 #48: Failed to handle RSP6: RSP6 missing ecdsa components on ThinkPad T460p (138a:0097):

Jamesxu182, 2018-09-22:
Prototype execution failed with error information __"Failed to handle RSP6: RSP6 missing ecdsa components, priv: 0, pub: 0"__

```
Prototype version 15
Found device 138a:0090

step 1
usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 64 81 08 9c 41 0c  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 07 
Expected at char 012
step 2
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 d0 cf c1 4b  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
step 3
usb write:
0000 43 02 
usb read:
0000 b0 04 
Expected len: 84, but got 2
step 4
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6 
01e0 31 28 dc 82 45 
usb read:
0000 00 00 
step 5
usb write:
0000 3e 
usb read:
0000 00 00 ef 00 40 00 00 10  01 00 00 01 01 00 05 00 
0010 01 04 07 00 00 10 00 00  00 10 00 00 02 01 02 00 
[... 242 lines cut ...]
0eb0 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0ec0 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0ed0 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0ee0 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0ef0 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0f00 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0f10 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0f20 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0f30 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0f40 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0f50 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0f60 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0f70 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0f80 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0f90 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0fa0 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0fb0 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0fc0 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0fd0 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0fe0 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0ff0 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
1000 ff ff ff ff ff ff ff ff 
Expected at char 008
Failed to handle RSP6: RSP6 missing ecdsa components, priv: 0, pub: 0
```

Jamesxu182, 2018-10-28:
The solution would be removing existing fingerprints from windows 10 (in a physical machine or virtual machine). Thanks a million

MartinX3, 2018-12-04:
Working on my T460P with Ubuntu. <3