nmikhailov/Validity90 #49: Add support for 06cb:008a

lakiluki1, 2018-10-18:
Hi, I'm using a Toshiba Portege X30D and it's driver hasn't been supported yet. I changed the following line in main.c (from the 06cb:009a driver) so I could run the prototype and provide the ouput:
`{ .vid = 0x06cb, .pid = 0x008a, .hasLed = 0, .hasBios = -1, .requiresReset = 0, .hasRawOutput = -1 }`
Currently, nothing works. The fingerprint sensor has no LEDs that can light up and the prototype doesn't wait for me to swipe my finger. Hopefully the output provides enough information to figure out what's going on.

[Prototype log](https://ptpb.pw/qLrm)

Supaiku, 2020-06-15:
I wonder if it might use a similar drive as 06cb:009a, which has a lot more attention: https://github.com/nmikhailov/Validity90/issues/34

the-muzammilali, 2026-10-15:
### Solution for Synaptics WBDI 06cb:008a (Viper family)

I have successfully reverse-engineered and enabled the `06cb:008a` sensor natively on Linux (Ubuntu 24.04 LTS / Toshiba Tecra X40-D / Portégé / Dynabook). 

#### Why previous attempts stalled:
1. **No Firmware Extension Exists or is Needed**: Unlike `0090`/`0097`/`009a`, the Viper `008a` ASIC (`57K0 FM-154-124`, ROM `6.20`) is strictly **Match-on-Host**. `GetFirmwareInfo (0x4302)` returns `0x04af` (no firmware detected), and MoC commands (`0x68`, `0x5e`) return error `0401` (invalid command). The official Windows driver (`synaWudfBioUsb.dll`) streams raw capacitive frames to the host CPU via UMDF.
2. **Flash Disconnects**: Attempting to write biometric templates to the SPI flash partition table causes bus contention and reboots (`USBError 19: No such device`).

#### How it works:
- **Frame Acquisition**: Calibrated using sensor init sequence and passively awaiting interrupts on EP `0x83` to stream full 56×144 8-bit grayscale matrix frames (8,064 bytes).
- **Matching Engine**: Host-side 2D Phase Correlation (`numpy.fft.fft2`) with 2D Hann window filtering (eliminating edge boundary artifacts) and an angular tilt search (`[-12° .. +12°]`) with threshold `0.12`.
- **System Integration**: Runs natively under `python-validity` + `open-fprintd` with template persistence in `/var/lib/python-validity/templates/`, fully enabling `fprintd-enroll`, `fprintd-verify`, and PAM authentication for `sudo` and lock screen.

#### Repository & Prebuilt .deb Packages:
- **Fork**: https://github.com/the-muzammilali/python-validity
- **Prebuilt Packages**: https://github.com/the-muzammilali/python-validity/releases/tag/v0.16-viper1
- **Upstream PR**: https://github.com/uunicorn/python-validity/pull/293

Install instructions:
```bash
sudo apt update
sudo apt install -y python3-usb python3-cryptography python3-numpy python3-pil python3-yaml python3-dbus python3-gi fprintd libpam-fprintd
sudo dpkg -i open-fprintd_0.6-viper1_all.deb python3-validity_0.16-viper1_all.deb
fprintd-enroll
fprintd-verify
```