nmikhailov/Validity90 #53: How to test on Lenovo Yoga? ID 06cb:0081 Synaptics, Inc.

shokuz, 2018-11-07:
Would love to test this. Is it in a usable state yet, or still proof of concept?

thebigfootsdad, 2019-04-11:
> Hi shokuz,
> 
> Yes, I have already test it.
> First I recommend you to create a folder to leave the project there.
> `mkdir protoFingerTest`
> 
> There do a git clone of the project:
> `git clone https://github.com/nmikhailov/Validity90.git`
> 
> Then in the prototype folder run the follow command:
> `cd prototype`
> `make`
> 
> Next give permission to the folder where the program is going to make the test of your FingerPrint files:
> `sudo chmod a+rwx /dev/bus/usb/<your path>:`
> For example in my computer it was "sudo chmod a+rwx /dev/bus/usb/001/005"
> 
> Finally go to the prototype folder and run the prototype program and save the information in a log.txt to share it with nmikhailov. I shared it opening a new issue.
> 
> `./prototype`
> 
> I hope is clear for you, any question ask me.
> 
> cheers!

When in the 'Prototype' folder, I used `make` in the terminal but got this result: 

```
Package nss was not found in the pkg-config search path.
Perhaps you should add the directory containing `nss.pc'
to the PKG_CONFIG_PATH environment variable
No package 'nss' found
Package openssl was not found in the pkg-config search path.
Perhaps you should add the directory containing `openssl.pc'
to the PKG_CONFIG_PATH environment variable
No package 'openssl' found
Package libusb-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusb-1.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libusb-1.0' found
Package libpng was not found in the pkg-config search path.
Perhaps you should add the directory containing `libpng.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libpng' found
Package glib-2.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `glib-2.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'glib-2.0' found
cc -c -Wall -g -DG_LOG_DOMAIN=\"Validity90\"  validity90/utils.c -o validity90/utils.o
validity90/utils.c:23:10: fatal error: gcrypt.h: No such file or directory
 #include <gcrypt.h>
          ^~~~~~~~~~
compilation terminated.
make: *** [Makefile:28: validity90/utils.o] Error 1
```

Do you know what the issue might be or how I could resolve it?

trueshanti, 2019-04-11:
Lenovo Yoga 910 : there is not much to see yet

./prototype 
Prototype version 15
Found device 138a:0094
Support would be available soon

TimLeijten, 2019-04-19:
install the dependencies specified in readme.md in prototype directory.

lesGrandsBras, 2019-06-02:
Hi!

```For example in my computer it was "sudo chmod a+rwx /dev/bus/usb/001/005"```
How do I know what path I have to give?

Thank you very much!

Kostanos, 2019-07-30:
In my case I just got this:
```
$ ./prototype 
Prototype version 15
Found device 06cb:0081
Support would be available soon
```

linux019, 2019-12-27:
Hi @nmikhailov I have the same issues, any luck to add support of `06cb:0081`?

raaz2053, 2020-01-13:
Prototype version 15
Found device 06cb:0081
Support would be available soon
:(

rahmatov, 2020-12-01:
Hi @nmikhailov , any news about support of 06cb:0081?