nmikhailov/Validity90 #57: Output from T480s (06cb:009a)

AnwariasEu, 2018-12-30:
I ran the prototype on a t480s with an 06cb:009a sensor.

To run the prototype I had to adjust the permissions for an usb-device `sudo chmod a+rw /dev/bus/usb/001/005`.

I hope the format is helpful.

https://termbin.com/bkas

```
Prototype version 15
Found device 06cb:009a

step 1
usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 91 d0 49 68 7c 49  00 23 00 00 00 00 01 00 
0020 00 00 02 d1 00 02 
Expected at char 012
Sensor not initialized, init byte is 0x2 (expected 0x02)
step 2
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 23 40 6b cd  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 23 40 6b cd  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
step 4
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6 
01e0 31 28 dc 82 45 
usb read:
0000 be 04 
[... 811 lines cut ...]
0360 00 00 00 00 00 00 00 00  00 00 00 00 00 
usb read:
0000 01 04 
Expected len: 298, but got 2
0000 01 04 
step 12
usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 91 d0 49 68 7c 49  00 23 00 00 00 00 01 00 
0020 00 00 02 d1 00 02 
Expected at char 011
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 91 d0 49 68 7c 49  00 23 00 00 00 00 01 00 
0020 00 00 02 d1 00 02 
step 13
usb write:
0000 1a 
usb read:
0000 91 04 
Expected at char 000
0000 91 04 
Incomplete reverse engineering
```

mlalkaka, 2019-01-01:
Regarding the permissions issue, I have published a pull request to fix this on Lenovo T480s laptops: #56. I'd be happy to address any feedback required to get it accepted :).

opatut, 2019-02-13:
I have another [output](https://gist.github.com/opatut/d3740bf12e6aae9ab23ae7245ee47f57).

I did not place my finger on the sensor, but it reports (at the end, after init, pressing "1", and a lot of hex output):

```
Awaiting fingerprint:
interrupt:
0000 00 00 00 00 00 
Waiting for finger...
interrupt:
0000 02 00 40 10 00 
Finger is on the sensor...
interrupt:
0000 03 40 01 00 00 
Scan in progress...
interrupt:
0000 03 41 03 00 40 
Fingerprint scan completed...
```

I'd love to help get this further, and willing to dig into code. Sadly I have no idea where I could start... so at least I can report this ;)

nunofranciscomoreira, 2019-04-09:
It didn't finish, and it says it recognizes the fingerprint on the sensor but it wasn't there.

Prototype version 15
Found device 06cb:009a

step 1
usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 c0 eb 89 68 5f d5  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 07 
Expected at char 012
step 2
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 bd 11 76 72  00 00 00 00 cc 39 7e 50 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
step 3
usb write:
0000 43 02 
usb read:
0000 00 00 01 00 02 00 08 00  28 4b 3f 5b 01 00 34 46 
0010 02 00 07 00 00 42 00 00  01 00 84 08 01 00 07 00 
0020 00 04 00 00 02 00 84 28  03 00 12 00 40 10 00 00 
0030 02 00 76 36 01 00 0c 00  10 0a 00 00 01 00 86 47 
0040 00 00 01 00 50 5a 00 00  02 00 23 77 00 00 01 00 
0050 80 2f 00 00 02 00 00 37  00 00 01 00 b0 08 02 00 
0060 01 00 25 56 00 00 01 00  60 04 00 00 
Expected len: 84, but got 108
step 4
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6 
01e0 31 28 dc 82 45 
usb read:
0000 be 04 
Expected at char 000
[... 1006 lines cut ...]
0890 00 00 01 00 00 00 02 00  00 00 00 00 f5 03 00 00 
08a0 a1 01 4e 00 00 00 00 00  00 00 e8 03 00 00 64 01 
08b0 00 00 d0 02 00 00 00 01  00 00 00 00 00 00 05 00 
08c0 00 00 00 00 00 00 05 00  00 00 00 00 00 00 00 00 
08d0 00 00 00 00 00 00 70 00  00 00 00 00 00 00 00 00 
08e0 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
08f0 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0900 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0910 00 00 00 00 00 00 02 00  00 00 00 00 00 00 00 00 
0920 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0930 00 00 00 00 00 00 00 00  00 00 00 00 00 00 
Awaiting fingerprint:
interrupt:
0000 00 00 00 00 00 
Waiting for finger...
interrupt:
0000 02 00 40 10 00 
Finger is on the sensor...
interrupt:
0000 03 40 01 00 00 
Scan in progress...
interrupt:
0000 03 41 03 00 40 
Fingerprint scan completed...
^C

maciejmotyka, 2020-02-16:
`Scan fingerprint` doesn't wait for the finger to be on the sensor.
I used `meld` to compare the output with and without my finger on, and there are vast differences, so I think that the scan works.

See the log here:
[sans-finger-T480s.log](https://github.com/nmikhailov/Validity90/files/4209571/sans-finger-T480s.log)