nmikhailov/Validity90 #69: 06cb:00a8 not detected

giovannicaligaris, 2019-05-05:
Clevo N151ZU

```
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 04f2:b685 Chicony Electronics Co., Ltd 
Bus 001 Device 003: ID 06cb:00a8 Synaptics, Inc. 
```

```
$ sudo make
cc -c -Wall -g -DG_LOG_DOMAIN=\"Validity90\" -I/usr/include/nss -I/usr/include/nspr -I/usr/include/libusb-1.0 -I/usr/include/libpng16 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include validity90/utils.c -o validity90/utils.o
cc -c -Wall -g -DG_LOG_DOMAIN=\"Validity90\" -I/usr/include/nss -I/usr/include/nspr -I/usr/include/libusb-1.0 -I/usr/include/libpng16 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include validity90/validity90.c -o validity90/validity90.o
cc -c -Wall -g -DG_LOG_DOMAIN=\"Validity90\" -I/usr/include/nss -I/usr/include/nspr -I/usr/include/libusb-1.0 -I/usr/include/libpng16 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -w main.c -o main.o
cc validity90/utils.o validity90/validity90.o main.o -o prototype  -lnss3 -lnssutil3 -lsmime3 -lssl3 -lplds4 -lplc4 -lnspr4 -lssl -lcrypto -lusb-1.0 -lpng16 -lz -lglib-2.0 -lgcrypt

$ sudo make permissions 
sudo chmod a+r /sys/class/dmi/id/product_serial
lsusb -d 138a: | awk -F '[^0-9]+' '{ print "/dev/bus/usb/" $2 "/" $3 }' | xargs -r sudo chmod a+rw
lsusb -d 06cb:0081 | awk -F '[^0-9]+' '{ print "/dev/bus/usb/" $2 "/" $3 }' | xargs -r sudo chmod a+rw
lsusb -d 06cb:009a | awk -F '[^0-9]+' '{ print "/dev/bus/usb/" $2 "/" $3 }' | xargs -r sudo chmod a+rw

$ sudo ./prototype 
Prototype version 15
No devices found
```

Am I do something wrong?

parkerlreed, 2019-05-05:
>Reverse engineering of Validity/Synaptics 138a:0090, 138a:0094, 138a:0097, 06cb:0081, 06cb:009a fingerprint readers protocol

008a isn't in the list...

nmikhailov, 2019-05-06:
@giovannicaligaris you can try to add it manually, its on line https://github.com/nmikhailov/Validity90/blob/master/prototype/main.c#L79 .

giovannicaligaris, 2019-05-06:
@nmikhailov sorry I'm a bit confused, what is it that you want me to do?

nmikhailov, 2019-05-06:
Add your device's vid:pid to the list in that file, then recompile and test
again.

giovannicaligaris, 2019-05-06:
I managed to get it to work, but in the end it says "incomplete reverse engineering".

Here's the log

`./prototype 
Prototype version 15
Found device 06cb:00a8

step 1
usb write:
0000 01 
usb read:
0000 00 00 10 b6 04 54 9a 00  00 00 06 14 01 33 00 01 
0010 00 00 b2 f7 89 00 6a ea  00 21 00 00 00 00 01 00 
0020 00 00 00 00 00 06 
Expected at char 002
Sensor not initialized, init byte is 0x6 (expected 0x02)
step 2
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 97 58 1c 34  00 00 00 00 fd cf d2 71 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 97 58 1c 34  00 00 00 00 fd cf d2 71 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
step 4
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6 
01e0 31 28 dc 82 45 
usb read:
0000 be 04 
Expected at char 000
0000 be 04 
step 5
usb write:
[... 810 lines cut ...]
0350 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0360 00 00 00 00 00 00 00 00  00 00 00 00 00 
usb read:
0000 01 04 
Expected len: 298, but got 2
0000 01 04 
step 12
usb write:
0000 01 
usb read:
0000 00 00 10 b6 04 54 9a 00  00 00 06 14 01 33 00 01 
0010 00 00 b2 f7 89 00 6a ea  00 21 00 00 00 00 01 00 
0020 00 00 00 00 00 06 
Expected at char 002
0000 00 00 10 b6 04 54 9a 00  00 00 06 14 01 33 00 01 
0010 00 00 b2 f7 89 00 6a ea  00 21 00 00 00 00 01 00 
0020 00 00 00 00 00 06 
step 13
usb write:
0000 1a 
usb read:
0000 91 04 
Expected at char 000
0000 91 04 
Incomplete reverse engineering`

giovannicaligaris, 2019-05-06:
By the way, this is what I added to main.c

`{ .vid = 0x06cb, .pid = 0x00a8, .hasLed = 1, .hasBios = 1, .requiresReset = 0, .hasRawOutput = 1 },
`
I got this information from

```
lsusb -v -d 06cb:00a8
Bus 001 Device 003: ID 06cb:00a8 Synaptics, Inc. 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass        16 
  bDeviceProtocol       255 
  bMaxPacketSize0         8
  idVendor           0x06cb Synaptics, Inc.
  idProduct          0x00a8 
  bcdDevice            1.54
  iManufacturer           0 
  iProduct                0 
  iSerial                 1 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0035
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           5
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval              10

```

nmikhailov, 2019-05-07:
@giovannicaligaris do you have windows or windows VM?
You can try passing though sensor to windows VM, installing official driver there and configuring it.
Then retest prototype again.

giovannicaligaris, 2019-05-10:
Should I run prototype while the Windows VM is running in the background?

giovannicaligaris, 2019-05-11:
@nmikhailov I did what you said but I keep getting the same result.

nmikhailov, 2019-05-11:
@giovannicaligaris so to confirm: you have configured sensor in windows VM and are able to log in with it in VM, but the prototype still doesn't work, right?

giovannicaligaris, 2019-10-12:
@nmikhailov sorry for the very late reply. I tried doing this through Virtualbox but it couldn't find the finger reader.

Maybe I am not doing it right. Is it possible you can guide me on how to do it?

casimirextreme, 2019-11-23:
> @giovannicaligaris do you have windows or windows VM?
> You can try passing though sensor to windows VM, installing official driver there and configuring it.
> Then retest prototype again.

I've the same device: Bus 001 Device 002: ID 06cb:00a8 Synaptics, Inc

The device has been configured from windows and is working properly but prototype is also ending with "Incomplete reverse engineering".

What else can I do to help?

Rio6, 2020-08-27:
I have 06cb:00a8 on my laptop too. I have a dual boot Windows 10 installation and have the scanner working on Windows. Can I do anything to help?

karthikiyengar, 2023-10-18:
Sorry to resurrect this old thread, but I was wondering if there's continued interest in this? I'd be happy to poke around and see whether we could make this work, if pointed in the right direction. For context, I'm on a Clevo N150CU (also resold as the System76 Darter Pro, Tuxedo InfinityBook v5).

freechelmi, 2023-10-19:
Yes pretty interested in supporting this OOTB and this is a really popular device in a lot of clevo models inluding , new NL41PU and NL55PU can includes it ( and many others )