nmikhailov/Validity90 #71: HP Spectre x360 (06cb:00bb)

Thalhammer, 2019-05-13:
I am working on a HP Spectre 13

Machine name: HP Spectre x360 Convertible 13-ap0xxx
Fingerprint is at 06cb:00bb, I edited both the makefile and prototype to add the pid, I copied the device entry of 06cb:009a.
Not sure if it is the correct on though.
<details>
     <summary>Prototype output</summary>
<pre>
Prototype version 15
Found device 06cb:00bb

step 1
usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 55 0c 4b 68 77 f1  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 07 
Expected at char 012
step 2
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 95 4f c8 97  00 00 00 00 9a 7e ff 0b 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
step 3
usb write:
0000 43 02 
usb read:
0000 00 00 01 00 02 00 08 00  da e9 07 5b 01 00 34 46 
0010 02 00 07 00 d0 3f 00 00  01 00 84 08 01 00 07 00 
0020 00 04 00 00 02 00 84 28  03 00 12 00 70 10 00 00 
0030 02 00 76 36 01 00 0c 00  10 0a 00 00 01 00 86 47 
0040 00 00 01 00 50 5a 00 00  02 00 23 77 00 00 01 00 
0050 80 2f 00 00 02 00 00 37  00 00 01 00 b0 08 02 00 
0060 01 00 25 56 00 00 01 00  60 04 00 00 
Expected len: 84, but got 108
step 4
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
[... 353 lines cut ...]
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval              10
Device Status:     0x0000
  (Bus Powered)
</pre>
</summary>

I would check myself but I don't know anything about those sensors, however, I am happy to do all sorts of tests if there is interest in supporting it.
If you need additional information let me know.

RaghavRao, 2019-05-13:
"Reverse engineering of Validity/Synaptics **138a:0090, 138a:0094, 138a:0097, 06cb:0081, 06cb:009a** fingerprint readers protocol"

06cb:00bb is out of scope for this project, you will most likely not receive much support for out of scope sensors

Thalhammer, 2019-05-13:
@RaghavRao I am aware that the sensor is not in scope for support, but I believe the protocol is similar, so at one day it might be. That's why I said, "if there is interest in supporting it".

If nobody decides to ever support it, I don't care. If someone decides to support it, the information above might help.

TO BE CLEAR: This is just here for the record. I don't use the sensor at all (not even on windows where it would work). I don't care if it never gets support.

AnterCreeper, 2022-03-02:
This type is specially for Hewlett Packard, and sometimes it needs intel sgx and hard to be reverse engineered.