nmikhailov/Validity90 #72: Add support for 06cb:00bd

mariobodemann, 2019-06-24:
Hey,

I hope I am not overlooking something, but currently I don't seem able to see the fingerprint sensor from the `Lenovo T490S` supported.

I ran

`lsusb | grep -i 06cb`
> Bus 001 Device 010: ID 06cb:00bd Synaptics, Inc. 

and build the prototype from master (8b80cfeae15487d5e13ff5ff3ab4f282eac765be):

`cd prototype; make; ./prototype`
> Prototype version 15
> No devices found

Can I somehow support you in implementing this reader?

JohnAZoidberg, 2019-06-24:
You can try to add your device [here](https://github.com/nmikhailov/Validity90/blob/master/prototype/main.c#L78) and see if it does something useful then.

mariobodemann, 2019-06-24:
Hey thanks, 

I added (inspired by the other devices with the same (v?)endor)
```
{ .vid = 0x06cb, .pid = 0x00bd, .hasLed = 0, .hasBios = -1, .requiresReset = 0, .hasRawOutput = -1 },
```

and get this result [prototype.log](https://github.com/nmikhailov/Validity90/files/3321528/prototype.log) ending on `Incomplete reverse engineering`. 

So I fear it didn't do the thing we want it to do?

t-8ch, 2019-06-27:
This should end up in one of the next proper versions of libfprint: https://gitlab.freedesktop.org/vincenth/libfprint/tree/synaptics-driver-20190617 (I think)

Ana06, 2019-07-07:
I also have a T490s, whose fingerprint doesn't work. What is the state of this? Something I can do to help?

pblack476, 2019-07-14:
Is the reader on the X1 Extreme among the ones being worked on? I have one and would like to know if I can offer any assistance whatsoever.

ab-pm, 2019-08-02:
According to [this forum response](https://forums.lenovo.com/t5/Other-Linux-Discussions/Linux-on-T495/m-p/4474320#M13440), Lenovo, Redhat and Synaptics are working on a driver.
Also it looks like someone started working on it in [this `libfprint` repository](https://gitlab.freedesktop.org/vincenth/libfprint/tree/synaptics-driver-20190617).

parkerlreed, 2019-08-02:
Here's to hoping this applies to the rest of the devices too

138a:0094 checking in

t-8ch, 2019-08-03:
> EDIT: :O https://gitlab.freedesktop.org/vincenth/libfprint/blob/synaptics-driver-20190617/libfprint/fprint-list-udev-rules.c#L30

This list has been there for some time and I don't think it gives any information about the currently ongoing work.

parkerlreed, 2019-08-06:
Ahh, poop. Thanks.

polatsinan, 2019-11-22:
On my ThinkPad T495s I have the following device:
`Bus 004 Device 006: ID 06cb:00bd Synaptics, Inc. 
`

I added the following to main.c, compiled and ran it:
`{ .vid = 0x06cb, .pid = 0x00bd, .hasLed = 0, .hasBios = -1, .requiresReset = 0, .hasRawOutput = -1 },
`

Result:
```

[...]

usb read:
0000 01 04 
Expected at char 000
0000 01 04 
Incomplete reverse engineering
```

Anything I can do?

bumbummen99, 2019-11-22:
@polatsinan As far as i am informed (got the T495 too) it should work with fprintd being ported to libfprint2 (https://gitlab.freedesktop.org/libfprint/libfprint/merge_requests/63/commits) which has just been merged.

Also note that a firmware update of the device will be required for that to work following to this discussion (https://gitlab.freedesktop.org/libfprint/libfprint/issues/181)

alxchk, 2020-02-15:
Just in case for those who cares - 
Works on latest libfprintd/fprintd + firmware

```
Bus 001 Device 003: ID 06cb:00bd Synaptics, Inc. 

> fprintd-verify 
Using device /net/reactivated/Fprint/Device/0
Listing enrolled fingers:
 - #0: right-index-finger
Verify result: verify-match (done)
```

FalcoSuessgott, 2020-02-26:
@alxchk 
Can you describe what u did to get it working?

bumbummen99, 2020-02-26:
@FalcoSuessgott You have to update the driver of your device (enable testing repositories to get the latest since it is not stable) and use the latest libfprint and fprintd (propably custom built since there is no up to date PPA).

Chrescht, 2020-04-09:
@bumbummen99 thanks, but could you be more specific, what libfprint version works?
I have currently version 1.90.1 and I keep getting errors about the device being already open:
Using device /net/reactivated/Fprint/Device/0
failed to claim device: Open failed with error: Device 06cb:00bd is already open

FreezFree, 2020-05-10:
Same problem here with the Version 1.90.1:
`$ fprintd-verify`
`Using device /net/reactivated/Fprint/Device/0
failed to claim device: Open failed with error: The driver encountered a protocol error with the device.`

Reinstall doesn't help.

_Thinkpad T495
Fedora release 32 (Thirty Two)_

jamescassell, 2020-05-10:
You need the firmware update from lvfs-testing.

bumbummen99, 2020-05-10:
There are issues with the current testing firmware that are fixed and it is being released currently.

mandresve, 2020-05-19:
Hello. There's suport for 06cb:00be | Synaptics? If not, how can I help?
I'm running Manjaro (Arch) latest ver.

charleswilmot, 2020-06-03:
Just reporting. Issue still occurs on

Fedora 32
Lenovo Carbon X1 gen 8
fprintd version 1.90.1

with firmware info

```
├─Prometheus:
│ │   Device ID:           d432baa2162a32c1554ef24bd8281953b9d07r11
│ │   Summary:             Fingerprint reader
│ │   Current version:     10.01.3121519
│ │   Vendor:              Synaptics (USB:0x06CB)
│ │   Install Duration:    2 seconds
│ │   Device Flags:        • Updatable
│ │                        • Cryptographic hash verification is available
│ │ 
│ └─Prometheus IOTA Config:
│       Device ID:         8dcffb5d059857368caa2b7b22371ebd831b0r0c
│       Current version:   0022
│       Minimum Version:   0022
│       Vendor:            Synaptics (USB:0x06CB)
│       Device Flags:      • Updatable
```

No newer firmware seem to be available after enabling 

```
fwupdmgr enable-remote lvfs-testing
fwupdmgr update
• Prometheus has no available firmware updates
```

vinayakumarb, 2020-07-08:
> Just reporting. Issue still occurs on
> 
> Fedora 32
> Lenovo Carbon X1 gen 8
> fprintd version 1.90.1
> 
> with firmware info
> 
> ```
> ├─Prometheus:
> │ │   Device ID:           d432baa2162a32c1554ef24bd8281953b9d07r11
> │ │   Summary:             Fingerprint reader
> │ │   Current version:     10.01.3121519
> │ │   Vendor:              Synaptics (USB:0x06CB)
> │ │   Install Duration:    2 seconds
> │ │   Device Flags:        • Updatable
> │ │                        • Cryptographic hash verification is available
> │ │ 
> │ └─Prometheus IOTA Config:
> │       Device ID:         8dcffb5d059857368caa2b7b22371ebd831b0r0c
> │       Current version:   0022
> │       Minimum Version:   0022
> │       Vendor:            Synaptics (USB:0x06CB)
> │       Device Flags:      • Updatable
> ```
> 
> No newer firmware seem to be available after enabling
> 
> ```
> fwupdmgr enable-remote lvfs-testing
> fwupdmgr update
> • Prometheus has no available firmware updates
> ```
> 
> EDIT: problem fixed on my system
> 
> I needed to update `fwupd` before updating the fingerprint reader's firmware
> Everything is working as a charm now

Thanks, Works fine on Ubuntu 20.4 now after said steps.

kleash, 2020-07-23:
@charleswilmot @vinayakumarb Can you please share exact steps you did to make it working.

I did following steps but `fprintd-verify` still throws same error:
`sudo apt-get update`
`sudo apt-get upgrade`
`fwupdmgr enable-remote lvfs-testing`
`fwupdmgr update`

charleswilmot, 2020-07-24:
@kleash same step as you did, but on a different OS.

Vozf, 2020-08-04:
For me it helped to reboot and do `fwupdmgr update`
After that login with fingerprint started working

amtux, 2020-08-08:
thanks for the heads up @Vozf.
I ran into the same issue where I enabled lvfs-testing and no still no updates were available but showed up after doing a reboot!

kleash, 2020-08-11:
@Vozf @amtux which linux you guys are using? bcoz it's not working for me in Ubuntu 20.04

Vozf, 2020-08-11:
Ubuntu 20.04 :)

amtux, 2020-08-11:
> @Vozf @amtux which linux you guys are using? bcoz it's not working for me in Ubuntu 20.04

Manjaro with Kernel version `5.7.9`

JAicewizard, 2020-08-14:
I just rebooted as that is supposed to get the updates to show up, but now fwupdmgr doesnt show the fingerprint reader at all.
Does anyone know how this would even be possible, or better yet, how to get it to show up again?

mickare, 2020-09-05:
Hi, thanks to everyone i got the fingerprint sensor `06cb:00bd` on my new Lenovo T14 working. :smile: 

System: Arch Linux 5.8.3-2-MANJARO

Steps:
1. (Optional ?) Upgrade Kernel to >= 5.8
1. [Install fwupdmgr](https://wiki.archlinux.org/index.php/Fwupd#Installation) via: `pacman -S fwupd`
2. Check if your device is listed: `fwupdmgr get-devices`, if not reboot
3. Enable "Linux Vendor Firmware Service" via:  `fwupdmgr enable-remote lvfs`
4. Refresh firmware list: `fwupdmgr refresh`
5. Run `fwupdmgr update` and update both Prometheus devices.
6. Reboot

Unfortunately `fingerprint-gui` does not detect the device; but you can either register your fingerprints via the system settings (Gnome: Settings > Users > Your User > Fingerprint Login); or use the fprintd command line: [man fprintd](https://jlk.fjfi.cvut.cz/arch/manpages/man/fprintd.1)

PS: Upgrade your kernel to at least 5.8 for the latest drivers for the AMD Ryzen 7 Pro 4750U; if not your CPU is ~50% slower.

amtux, 2020-09-05:
@mickare - agree. i had a bunch of issues with 5.7.x on T14s with Ryzen 7 4750U, stuff like mic not working and random GPU crashes when external monitor was plugged in via DP/HDMI. Linux 5.8 fixed all the issues and it's running ever more stable on 5.8.3 right now.

JAicewizard, 2020-09-06:
5.8 made a lot of things better on my T14, I think the no mic was actually listed on the arch wiki.

Punkoivan, 2020-09-30:
@mickare thank you, now sensor works on my t14 even without reboot.

Folks, may be we can close the issue?

victortrac, 2020-09-30:
I think this can be closed.

sreehax, 2020-10-02:
I'm on Void Linux on my ThinkPad X1 Carbon Gen 7 and I tried all the steps with enabling lvfs-testing and rebooting and all but it still says
`list_devices failed: No devices available`

sreehax, 2020-10-02:
I have `Bus 001 Device 006: ID 06cb:00bd Synaptics, Inc. ` in lsusb

caj-larsson, 2021-02-05:
This sensor works for me for an hour to a day before i can no longer verify any part of my finger, how can I debug this? Is this a sensitivity issue or potentially noise in the images matched? While I know my finger print does change over time, I know the sensor works fine in windows so it definitely seem to be an issue with my system or this driver

matiaslopezd, 2021-02-11:
The way to fix this is to reboot your laptop, log in, open the terminal and enter:

`fwupdmgr update`

Then reboot (will be part of update) and you will available to use the fingerprint. :smile:

Vicsuka, 2021-02-11:
`ID 06cb:00dc Synaptics, Inc. ` Ubuntu 20.04.2 LTS update doesn't do anything for me

Vicsuka, 2021-02-11:
`├─Touchpad:
│     Current version:     1.2.3078276
│     Bootloader Version:  8.0
│     Vendor:              Synaptics (HIDRAW:0x06CB)
│     Device Flags:        • Internal device
│                          • Updatable
`
Ubuntu 20.04.2 LTS, update doesn't work. Anyone else having the same device?

matiaslopezd, 2021-02-11:
@Vicsuka update to 20.10, I had a lot of [issues](https://discourse.ubuntu.com/t/psa-20-04-2-install-failure-bug-images-soon-to-be-recreated/20846?u=popey) inclusive in an empty disk installation of 20.04.2.

Photon89, 2021-08-14:
> Unfortunately fingerprint-gui does not detect the device; but you can either register your fingerprints via the system settings (Gnome: Settings > Users > Your User > Fingerprint Login); or use the fprintd command line: man fprintd

This solved it for me, thanks @mickare!