nmikhailov/Validity90 #73: Add support for 06cb:00be

Ifnnx, 2019-06-27:
It would be really nice, if i could use my fingerprintreader in my new lenovo Ideapad. ;)

xeu100, 2020-01-25:
Any progress on this?

jsplate, 2020-05-02:
This project seems not quite active.
There has been a little bit of hope for **20.04 Focal Fossa**:
https://askubuntu.com/questions/1214592/fingerprint-reader-lenovo-thinkpad-l13-synaptics-driver

pfuhlert, 2020-05-11:
Anyone got it to work with 20.04?

xeu100, 2020-05-11:
I have not got it working yet.

mandresve, 2020-05-19:
I'll test it today. "You'll need at least fwmgr-1.3.5 and fprintd-1.90. Then enable testing remote in fwupdmgr. See here how: github.com/fwupd/fwupd/wiki/LVFS-Testing-remote Firmware you need to to install is listed here: fwupd.org/lvfs/search?value=prometheus – zeroinfinity

p-try, 2020-05-20:
It didn't work for me. The firmware from the link is for 00bd only according to the contained XML file.

xeu100, 2020-05-22:
It does not support this device, unfortunately.

Freedom9339, 2020-06-09:
I've added $100 bounty on https://www.bountysource.com/issues/76164422-add-support-for-06cb-00be for anyone who can solve it. I need this.

Dani9OO, 2020-09-02:
I too would like to see this supported +1

sherer-pro, 2020-09-03:
I support, really need

quaxam, 2020-09-06:
Support too, really need it soon :( So many advances in 00bd but not for our oobe

ADernild, 2020-12-29:
I too would like to see this supported ☺️

suchipi, 2021-02-23:
I added $250 to the bounty for this (non-expiring), so the bounty for this issue is now at $350.

nikolobok, 2021-02-23:
> I added $250 to the bounty for this (non-expiring), so the bounty for this issue is now at $350.
It wouldn`t be like buy a pancake on the street for some digits with any symbols.. rather like janitor or porter, do you ready to let him in?

Popax21, 2021-06-14:
It's been a while, but I'm currently working on reverse engineering the Windows driver for this model (Synaptics FM-3463 / 06cb:00be). Don't expect anything soon though, as the driver is complex and it's more a side project than my current focus. I'll post major updates on progress here (if that's OK).

kmanwar89, 2021-07-15:
Just purchased a Lenovo Ideapad 5 Flex (Model 14ITL05) and everything on Linux works except the fingerprint reader!

I'm on Mint 20 Uma, which uses Ubuntu 20.04 as it's base. Kernel is 5.11.0-22-generic and I have a 06cb:00be Synaptic device shown in the output of _lsusb_.

If this shows any potential means of gaining traction, I may just increase the bounty to $500 total.

Popax21, 2021-07-30:
For anyone wondering how I'm progressing: I'm almost at the point of having fully functioning initialization & pairing routines. My reverse engineering effort started completly from scratch, so I didn't look much at the work inside this repo, but as it turns out, the tudor family of sensors (that's what they're called in the driver) shares some, but not all characteristics with the sensors which this repo focuses on. However, I was able to clarify many things, because I am reverse engineering the Windows driver (which conveniently has debug symbols/printfs inside of it), while (I think) they only looked at captured USB packets. I might create a pull request claryfing overlapping info. As of yet it's still to early to release anything, but when I do, I'll post the link (and any major updates on progress) here.

PS: Just speculation, but I belive that my driver could also work for 06cb:00bd sensors, without the need for a FW update. It might also possible that other sensors could work too, but so far I only tested it on my laptop's 06cb:00be one.

shaqman, 2021-07-31:
> For anyone wondering how I'm progressing: I'm almost at the point of having fully functioning initialization & pairing routines. My reverse engineering effort started completly from scratch, so I [...]

Sounds a step in the right direction.
I have the 06cb:00be sensor, so if you want me to confirm how it works here, I'm open to testing it. Just be adviced that I'm kinda new in this, so I might need more than the usual pointers if you are willing to.

NoPreserveRoot, 2021-07-31:
> For anyone wondering how I'm progressing: I'm almost at the point of having fully functioning initialization & pairing routines. My reverse engineering effort started completly from scratch, so I [...]

I'm also willing to be a guinea pig. I have an Ideapad Flex 5 (has 06cb:00be) with Arch that I can test it on. Like @shaqman, I have never been a driver tester so I may need a bit more instruction. Glad you're making progress, though!

Endermen1094, 2021-08-01:
> I'm also willing to be a guinea pig. I have an Ideapad Flex 5 (has 06cb:00be) with Arch that I can test it on. Like @shaqman, I have never been a driver tester so I may need a bit more instruction. [...]
I am too ok being a guinea pig.

patbakdev, 2021-08-02:
Count me in for testing as well. I have a Lenovo IdeaPad S740 (06cb:00be Synaptics, Inc.) running Manjaro and 5.13 kernel

Popax21, 2021-08-03:
Well that was quick. A few days after getting initialization to work, I already have events (finger press/remove, etc) and image captures up and running. Yet I'm still not done. Image post processing and matching is still missing (the sensor has commands for that functionality). After I get that to run, I'll put everything in a Github repo and send the link here.

I'll probably release the driver in two stages: First, I'll release my prototype python driver, too check if it works on sensors in the wild and too fix any bugs. After that, I'll create a libfprint implementation, so that it can work with existing fingerprint infrastructure.

PS: Here a little sneak peak: a capture of the side of my finger
[image]

mandresve, 2021-08-03:
Same as @patbakdev. Count me in for testing!. I have two Lenovo machines with 06cb:00be sensors from Synaptics. Also running Manjaro and 5.14.rc3 kernel.

pfuhlert, 2021-08-03:
Hey that sounds very nice! I can also test if you need another one 👍 Currently I'm running on a Thinkpad L490 with the 06cb:00be sensor using Ubuntu 21.04

Popax21, 2021-08-05:
I just uploaded my current state of progress: https://github.com/Popax21/synaTudor. Note that some things are still incomplete, like on sensor matching, or a libfprint integration, but it should still be enough to do some first in-the-wild tests.

PS: I'm beginning to loose motivation for this project, so I might take a break from it. So if you don't hear anything from me for a while, that's the reason.

mandresve, 2021-08-06:
Great effort @Popax21 It is working for me :smile: (Manjaro and 5.14.rc3 kernel) Thanks!
I look forward to hearing from you soon.

patbakdev, 2021-08-06:
Yes, this is great work. I had no problems. IdeaPad S740 with 5.10.53-1-MANJARO.

Popax21, 2021-09-05:
I've just uploaded a (primitive) libfprint integration, which just calls the python driver library (because I'm not in the mood of converting it to C as of now). However, don't get your hopes up, because I discovered that without the post processing algorithm which I yet have to reverse engineer, libfprint produce successfull matches, meaning: you can enroll, but not verify. I'm going to continue reverse engineering and posting updates here.

Popax21, 2021-09-06:
I figured out a way to extract the post processing code from the Windows driver and integrate it into the Python driver. However, libfprint is still unable to produce a match, even with the post processed images. I suspect that I have to implement on device matching, however that will take a while. Still, you should be able to test the libfprint integration + post processing code and check if it might work for you.

Etaash-mathamsetty, 2021-09-07:
I am getting a "did you setup the pairing data?"  error when running fprintd-enroll, are there an obvious errors that I could have made?

Etaash-mathamsetty, 2021-09-07:
I fixed it! And I was able to enroll my right index finger

patbakdev, 2021-09-07:
I wasn't able to make this work. Took some doing to get everything needed to build libfprint, etc., but then I ran into this issue. https://github.com/3v1n0/libfprint/issues/29.  

After downgrading fprintd to 1.90.1 and running fprintd-enroll I got:

```
$ list_devices failed: Failed to activate service 'net.reactivated.Fprint': timed out (service_start_timeout=25000ms)
# list_devices failed: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

systemd[1]: Starting Fingerprint Authentication Daemon...
fprintd[20880]: Traceback (most recent call last):
fprintd[20880]:   File "wrapper.py", line 1, in <module>
fprintd[20880]: ModuleNotFoundError: No module named 'tudor'
fprintd[20880]: Couldn't load wrapper python module!
systemd[1]: fprintd.service: Main process exited, code=dumped, status=5/TRAP
systemd[1]: fprintd.service: Failed with result 'core-dump'.
systemd[1]: Failed to start Fingerprint Authentication Daemon.
```

Probably something I am missing wrt setting everything up. I'll have to try again later when I have more time. 

Keep up the good work.

Popax21, 2021-09-07:
> I wasn't able to make this work. Took some doing to get everything needed to build libfprint, etc., but then I ran into this issue. [3v1n0/libfprint#29](https://github.com/3v1n0/libfprint/issues/29).
> 
> After downgrading fprintd to 1.90.1 and running fprintd-enroll I got:
> 
> ```
> $ list_devices failed: Failed to activate service 'net.reactivated.Fprint': timed out (service_start_timeout=25000ms)
> # list_devices failed: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
> 
> systemd[1]: Starting Fingerprint Authentication Daemon...
> fprintd[20880]: Traceback (most recent call last):
> fprintd[20880]:   File "wrapper.py", line 1, in <module>
> fprintd[20880]: ModuleNotFoundError: No module named 'tudor'
> fprintd[20880]: Couldn't load wrapper python module!
> systemd[1]: fprintd.service: Main process exited, code=dumped, status=5/TRAP
> systemd[1]: fprintd.service: Failed with result 'core-dump'.
> systemd[1]: Failed to start Fingerprint Authentication Daemon.
> ```
> 
> Probably something I am missing wrt setting everything up. I'll have to try again later when I have more time.
> 
> Keep up the good work.

Did you install the python driver module globaly? (see README.md in the pydrv directory)

Etaash-mathamsetty, 2021-09-24:
Any updates for the on device processing?
I think it might be some dumb mistake in the code elsewhere lol

Popax21, 2021-09-24:
> Any updates for the on device processing?
> I think it might be some dumb mistake in the code lol

As already said, I think I have to reverse engineer & replicate on-device matching for it to work, and that will take a while.

Popax21, 2021-10-14:
> could you give another update?

I'm still making progress, but it's quite slow. Hopefully though, the next update will be an announcement of the fully functioning driver.

Etaash-mathamsetty, 2021-11-03:
another progress update plz?

Popax21, 2021-11-03:
> another progress update plz?
I currently don't have much time or motivation to work on this project, however, when I regain both, I'll continue working on it.

Etaash-mathamsetty, 2021-11-04:
> > another progress update plz?
> 
> I currently don't have much time or motivation to work on this project, however, when I regain both, I'll continue working on it.

That's alright, 
Can you upload what you have got? I would like to continue working on it

Popax21, 2021-11-04:
> > > another progress update plz?
> > 
> > 
> > I currently don't have much time or motivation to work on this project, however, when I regain both, I'll continue working on it.
> 
> That's alright, but Can you upload what you have got? I would like to continue working on it

I didn't get much past the current state of the repo, so everything should be in there. However, most of the reverse engineering work is quite messy, so it may take a while until you get used to it. rev.txt and the Ghydra export are the most important files, as they contain all the information already gathered, and the prototype driver contains the practical implementation of it, which might be easier to understand.

Etaash-mathamsetty, 2021-11-05:
> I didn't get much past the current state of the repo, so everything should be in there. However, most of the reverse engineering work is quite messy, so it may take a while until you get used to it. [...]

good to know

Etaash-mathamsetty, 2021-11-07:
> I didn't get much past the current state of the repo, so everything should be in there. However, most of the reverse engineering work is quite messy, so it may take a while until you get used to it. [...]

I tried compiling your fork of ghidra and... it didn't work
any help?
im using arch linux now
error:
Could not find method helpIndexCompile() for arguments [javax.help:javahelp:2.0.05] on object of type org.gradle.api.internal.artifacts.dsl.dependencies.DefaultDependencyHandler.

Popax21, 2021-11-07:
> I tried compiling your fork of ghidra and... it didn't work any help? im using arch linux now error: Could not find method helpIndexCompile() for arguments [javax.help:javahelp:2.0.05] on object of [...]

Did you follow the regular Ghdira build guide?

Etaash-mathamsetty, 2021-11-07:
yes i did

Popax21, 2021-11-07:
> yes i did

I did some research, and it might be related to not building with JDK 11

Etaash-mathamsetty, 2021-11-07:
> > yes i did
> 
> I did some research, and it might be related to not building with JDK 11

wait JDK do I use then???

Popax21, 2021-11-07:
> > > yes i did
> > 
> > 
> > I did some research, and it might be related to not building with JDK 11
> 
> wait what JDK do I use then???

Try JDK 11

Etaash-mathamsetty, 2021-11-07:
> > > > yes i did
> > > 
> > > 
> > > I did some research, and it might be related to not building with JDK 11
> > 
> > 
> > wait what JDK do I use then???
> 
> Try JDK 11

That's what I am using though?

Etaash-mathamsetty, 2021-11-07:
also extra error
I removed the assert for it because I thought it was just some newer version
assert(sourceSha256.equals(expectedSha256))
         |            |      |
         |            false  '81b7d19d57c4a3009f4761699a72e8d642b5e1d9251d2bb98df438b1e28f8ba9'
         'e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855'

Etaash-mathamsetty, 2021-11-07:
I will try using regular ghidra for now I guess

Etaash-mathamsetty, 2021-12-03:
> > another progress update plz?
> 
> I currently don't have much time or motivation to work on this project, however, when I regain both, I'll continue working on it.

Progress update?

Etaash-mathamsetty, 2021-12-09:
I managed to disassemble the file in IDA freeware. 
I think this function matches the image
[func.txt](https://github.com/nmikhailov/Validity90/files/7689050/func.txt)
Unfortunately I don't know much assembly and I have no idea how to convert the file to C.
maybe one of you might find it useful.

Etaash-mathamsetty, 2021-12-10:
I tried to use the driver again but it failed with this error (the libfprint is too old :( )
usr/lib/fprintd: symbol lookup error: /usr/lib/fprintd: undefined symbol: fp_device_suspend_finish

nicman23, 2021-12-10:
@Etaash-mathamsetty you do use arch ?

Etaash-mathamsetty, 2021-12-10:
yes

nicman23, 2021-12-11:
[here](https://transfer.sh/OjwgiF/tudor.tar)  the fprintd is an older from the repos the libfprint includes the patch and python driver.

nicman23, 2021-12-11:
also here is the aur package https://aur.archlinux.org/packages/libfprint-tudor (a built version is in the above tar)

mandresve, 2021-12-11:
Hello and thanks for the latests efforts. 
I was testing the provided packages and everything goes just fine, but the fprintd-verify function always return verify-no-match for me after enrolling. Is this already supported? (Matching)
Thank you again.

Dani9OO, 2021-12-11:
> the fprintd-verify function always return verify-no-match for me after enrolling. Is this already supported?

It's not yet supported the following comments by Popax21 provide more info about it [1](https://github.com/nmikhailov/Validity90/issues/73#issuecomment-913855572) [2](https://github.com/nmikhailov/Validity90/issues/73#issuecomment-926594526)

Popax21, 2021-12-13:
Little status update: Nothing much happened behind the scenes since the last update, as I just never found the motivation to continue reverse engineering, mainly because that part of the driver references a lot of new submodules. I don't want to promise too much, but I think I am going to have more time/motivation to work on this over newyears.

[94 more replies]