nmikhailov/Validity90 #83: ubuntu 19.04 instalation problem bash: ./configure: No such file or directory

grekpg, 2019-09-10:
grek@grek-ThinkPad-A475:~/Validity90/libfprint$ sudo apt-get install make gcc libgcrypt-dev libglib2.0-dev libnss3-dev libusb-1.0-0-dev libssl-dev libpng-dev libgnutls28-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libgcrypt20-dev' instead of 'libgcrypt-dev'
gcc is already the newest version (4:8.3.0-1ubuntu3).
libgcrypt20-dev is already the newest version (1.8.4-3ubuntu1).
libpng-dev is already the newest version (1.6.36-6).
libusb-1.0-0-dev is already the newest version (2:1.0.22-2).
make is already the newest version (4.2.1-1.2).
libglib2.0-dev is already the newest version (2.60.4-0ubuntu0.19.04.1).
libgnutls28-dev is already the newest version (3.6.5-2ubuntu1.1).
libnss3-dev is already the newest version (2:3.42-1ubuntu2.1).
libssl-dev is already the newest version (1.1.1b-1ubuntu2.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

grek@grek-ThinkPad-A475:~/Validity90/libfprint$ ./configure
bash: ./configure: No such file or directory

grek@grek-ThinkPad-A475:~/Validity90/libfprint$ make
make: *** No targets specified and no makefile found.  Stop.
grek@grek-ThinkPad-A475:~/Validity90/libfprint$ make permissions
make: *** No rule to make target 'permissions'.  Stop.
grek@grek-ThinkPad-A475:~/Validity90/libfprint$ ./prototype
bash: ./prototype: No such file or directory

JohnAZoidberg, 2019-09-10:
There's no `configure`. I'm not sure what you're doing and where you got the instructions from.
If you want to try out the prototype, please follow the instructions [here](https://github.com/nmikhailov/Validity90/tree/master/prototype)

grekpg, 2019-09-10:
ok i found first ./autogen.sh then configure make make install, 
I installl also prototype here is my results : https://github.com/nmikhailov/Validity90/issues/82 (second post) 
Please tell my so this work ? when i run ./prototype run scan - fingeprinter start and work so can i use it on my system ? 

fprintd-enroll still give my 
list_devices failed: No devices available
 
so how i can setup authentification in ubuntu 19.04 kde ? is possible now?

galenasphaug, 2019-09-11:
The prototype is all there is for 138a:0097. It needs more development in order to work with libfprintd.