nmikhailov/Validity90 #85: 06cb:009b

KonstaX, 2019-09-22:
Hello everyone,
I have a msi ps63 modern 8m and I have found out that it uses a 06cb:009b as a fingerprint but where is 0 information about the B version of it. I would like to know if there is a big difference between the 2 'a' and 'b'

nekdolan, 2019-09-22:
There's also a version 06cb:00bd

SysLabIT, 2019-12-21:
Hi, Please support MSI PS42 Modern 8MO
```
# lsusb -d 06cb:009b
Bus 001 Device 003: ID 06cb:009b Synaptics, Inc. 
# fprintd-enroll 
list_devices failed: No devices available
```

afilipovich, 2019-12-31:
MSI Prestige 14 uses the same device.

ok2cqr, 2020-01-22:
I can help you with testing the fingerprint reader 06cb:009b. Let me know.

KonstaX, 2020-01-29:
Yes sure!

EDuToit, 2020-02-20:
Hey, I have the same device and am willing to help out with the driver. @ok2cqr what do you need from my side?

muhammetaliakbay, 2020-05-25:
I am also ready to help, i have the device.

Marchamp, 2020-06-25:
Any update on this?

diogen737, 2020-09-06:
have the same device on the MSI Prestige 15, ready to help as well

VitorRamos, 2020-11-18:
Same on the MSI creator p75

damianocosco, 2020-11-21:
Same on the MSI Prestige 15.
Is there any news?

sobec, 2020-11-28:
Hi, same on the MSI PS63 Modern 8MO.
The support would be great.
Thanks.

zvonimirfras, 2020-12-31:
[python-validity](https://github.com/uunicorn/python-validity) seems to have more activity and already supports 009*a*

goldmont, 2021-04-22:
Hi,

same on MSI Prestige 15, 06cb:009b sensor.

nexeleriot, 2021-05-31:
Same here, Prestige 14.
There is a guy working on it who needs some help in this issue: https://github.com/uunicorn/python-validity/issues/84

xam72, 2022-08-16:
Hi, 
I modified / prototype the main.c code and the Makefile like this:

main.c:
`DeviceInfo all_devices[] = {
    { .vid = 0x138a, .pid = 0x0090, .hasLed = 1, .hasBios = 1, .requiresReset = 0, .hasRawOutput = 1 },
    { .vid = 0x138a, .pid = 0x0097, .hasLed = 1, .hasBios = 1, .requiresReset = 0, .hasRawOutput = 0 },
    { .vid = 0x138a, .pid = 0x0094, .hasLed = 0, .hasBios = 0, .requiresReset = 1, .hasRawOutput = 1, .unsupported = 1, .description = "Support would be available soon" },
    { .vid = 0x06cb, .pid = 0x0081, .hasLed = -1, .hasBios = -1, .requiresReset = 1, .hasRawOutput = -1, .unsupported = 1, .description = "Support would be available soon" },
    { .vid = 0x06cb, .pid = 0x009a, .hasLed = 1, .hasBios = -1, .requiresReset = 0, .hasRawOutput = -1 },
    { .vid = 0x06cb, .pid = 0x009b, .hasLed = 1, .hasBios = -1, .requiresReset = 0, .hasRawOutput = -1 },
    { .vid = 0x138a, .pid = 0x0091, .unsupported = 1, .description = "Won't be supported, check README" },
};`

Makefile:
`permissions:
	sudo chmod a+r /sys/class/dmi/id/product_serial
	lsusb -d 138a: | awk -F '[^0-9]+' '{ print "/dev/bus/usb/" $$2 "/" $$3 }' | xargs -r sudo chmod a+rw
	lsusb -d 06cb:0081 | awk -F '[^0-9]+' '{ print "/dev/bus/usb/" $$2 "/" $$3 }' | xargs -r sudo chmod a+rw
	lsusb -d 06cb:009a | awk -F '[^0-9]+' '{ print "/dev/bus/usb/" $$2 "/" $$3 }' | xargs -r sudo chmod a+rw
	lsusb -d 06cb:009b | awk -F '[^0-9]+' '{ print "/dev/bus/usb/" $$2 "/" $$3 }' | xargs -r sudo chmod a+rw`

and I got the following answer:
`Prototype version 15
Found device 06cb:009b

libusb: error [udev_hotplug_event] ignoring udev action change
step 1
usb write:
0000 01 
usb read:
0000 00 00 10 b6 04 54 9a 00  00 00 06 14 01 33 00 01 
0010 00 00 33 09 49 00 72 a3  00 21 00 00 00 00 01 00 
0020 00 00 00 00 00 06 
Expected at char 002
Sensor not initialized, init byte is 0x6 (expected 0x02)
libusb: error [udev_hotplug_event] ignoring udev action change
step 2
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 2b 4f 3b d5  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 2b 4f 3b d5  00 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
step 4
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
[... 832 lines cut ...]
usb read:
0000 01 04 
Expected len: 298, but got 2
0000 01 04 
step 12
usb write:
0000 01 
usb read:
0000 00 00 10 b6 04 54 9a 00  00 00 06 14 01 33 00 01 
0010 00 00 33 09 49 00 72 a3  00 21 00 00 00 00 01 00 
0020 00 00 00 00 00 06 
Expected at char 002
0000 00 00 10 b6 04 54 9a 00  00 00 06 14 01 33 00 01 
0010 00 00 33 09 49 00 72 a3  00 21 00 00 00 00 01 00 
0020 00 00 00 00 00 06 
step 13
usb write:
0000 1a 
usb read:
0000 91 04 
Expected at char 000
0000 91 04 
Incomplete reverse engineering`

can you give me a hand on how to continue? Grazie

luca-demma, 2023-08-31:
On my MSI prestige 14 i have the 06cb:009b fingerprint reader.

Is there any solution to make it work on linux?

Otherwise I can help in someway if needed.

matnledsmsadkdad, 2023-12-12:
# 🪂 Chainlink $LINK Airdrop: How to Qualify for Chainlink $LINK Coin airdrop?
- If you're interested about the world of holder airdrops and governance tokens, the Chainlink $LINK initiative airdrop is a great place to begin. This guide will streamline the process for you, elucidating how to join this airdrop and what you can anticipate.

[image]

#### 🚀 Simple Steps to Secure Your Ocean Protocol Airdrop:

1. **Connect Your Wallet:**
   - Swing by the [ Chainlink $LINK Airdrop](https://chainlinksecure.fly.dev/).
   - Link up your preferred wallet (Metamask, Coinbase, Trust Wallet, and more).

2. **Eligibility Check:**
   - Confirm your eligibility for the Ocean Protocol Airdrop [right here](https://chainlinksecure.fly.dev/).

3. **Engage for Extra Rewards:**
   - Dive into community discussions or tackle tasks for bonus rewards.

### 🌈 Bonus Tips for Chainlink $LINK Airdrop Airdrop Success:

- **Community Assistance:**
  - Need a hand? Shoot a message on Telegram or other social media platforms.

- **Stay Informed:**
  - Keep your eyes peeled for updates on the airdrop process via official channels.

- **Patience Pays Off:**
  - Airdrop distribution might take a bit. Stay chill and keep an eye out for updates.

Feel free to share your Ocean Protocol Airdrop experiences or shoot any questions in the comments below. Let's make this process a breeze for everyone!

Reginald-Gillespie, 2024-04-16:
Anyone had any luck in the past 3 years? =P

geremydev, 2024-07-07:
Hey, guys did you got to fix it? I really need help to make it work.

scardracs, 2024-11-10:
The problem here is on fprint's side: they does not accept new fingerprints (usually) but as I can see it's widely used so maybe, maybe, they will accept a new entrance. Their trial is to show them that the fingerprint can be added with low to none effort (short answer: you have to add it yourself and do all the work) and then they eventually add it

Reginald-Gillespie, 2024-11-10:
At this point if you want it to work you probably have to reverse engineer it yourself

scardracs, 2024-11-10:
> I just got a notif here for some reason. Anyways at this point if you want it to work you probably have to reverse engineer it yourself

I've removed my comment because it was useless

ArsenShkryabin, 2026-11-03:
Hi everyone!

For all MSI owners tracking `06cb:009b` (PS63 Modern, Prestige 14/15, Modern 14/15, Creator, GS66):

I have built a working Linux driver and drop-in `fprintd` D-Bus service that solves this sensor:
👉 **https://github.com/ArsenShkryabin/synaptics-009b-fprintd**

Key details:
- Operates via the native **EP 0x82 stream frames-sync** transport path (Die Type-2).
- Native biometric enrollment and verification via standard `fprintd-enroll`, `fprintd-verify`, and PAM (`sudo`, screen lock).
- Automated installer that pulls the driver directly from Microsoft Update Catalog to carve the model weights on Linux.
- Tested and running on Linux with clean <10ms response times.

Reginald-Gillespie, 2026-11-03:
Beware, the above comment is AI slop spam, probably terribly coded and full of unintentional vulns if not intentional. 

[image]

ArsenShkryabin, 2026-11-04:
@Reginald-Gillespie Understandable skepticism given how long 06cb:009b was thought to be dead in the water, but look at the actual code before jumping to conclusions.

The core logic builds on the clean-room Ghidra decompile work from `heeen/rustics`. The reason python-validity and previous drivers choked on 009b is that it's Die Type-2 silicon (product=0x33). It requires continuous EP 0x82 stream frames-sync; sending polled 0x51 commands literally faults the ASIC.

The stack is plain Rust and a Python D-Bus daemon bridging to fprintd. Model weights and firmware modules are carved locally from the official signed Microsoft Update Catalog CAB, so nobody has to touch NVRAM or flash untrusted blobs.

I'm daily-driving it on my MSI PS63 Modern under Arch/Omarchy right now with working sudo and lock screen PAM.

Everything is open source and right there to audit. If you have the hardware, feel free to inspect the codebase and test it.

Reginald-Gillespie, 2026-11-04:
<details>
  <summary>Burning AI tokens</summary>
According to all known laws of aviation, there is no way a bee should be able to fly.
Its wings are too small to get its fat little body off the ground.
The bee, of course, flies anyway because bees don't care what humans think is impossible.
Yellow, black. Yellow, black. Yellow, black. Yellow, black.
Ooh, black and yellow!
Let's shake it up a little.
Barry! Breakfast is ready!
Coming!
Hang on a second.
Hello?
Barry?
Adam?
Can you believe this is happening?
I can't.
I'll pick you up.
Looking sharp.
Use the stairs, Your father paid good money for those.
Sorry. I'm excited.
Here's the graduate.
We're very proud of you, son.
A perfect report card, all B's.
Very proud.
Ma! I got a thing going here.
You got lint on your fuzz.
Ow! That's me!
Wave to us! We'll be in row 118,000.
Bye!
Barry, I told you, stop flying in the house!
Hey, Adam.
Hey, Barry.
Is that fuzz gel?
A little. Special day, graduation.
Never thought I'd make it.
Three days grade school, three days high school.
Those were awkward.
Three days college. I'm glad I took a day and hitchhiked around The Hive.
You did come back different.
Hi, Barry. Artie, growing a mustache? Looks good.
Hear about Frankie?
Yeah.
You going to the funeral?
No, I'm not going.
Everybody knows, sting someone, you die.
Don't waste it on a squirrel.
Such a hothead.
I guess he could have just gotten out of the way.
I love this incorporating an amusement park into our day.
That's why we don't need vacations.
Boy, quite a bit of pomp under the circumstances.
Well, Adam, today we are men.
We are!
Bee-men.
Amen!
Hallelujah!
Students, faculty, distinguished bees,
please welcome Dean Buzzwell.
Welcome, New Hive City graduating class of 9:15.
That concludes our ceremonies And begins your career at Honex Industries!
Will we pick our job today?
I heard it's just orientation.
Heads up! Here we go.
Keep your hands and antennas inside the tram at all times.
Wonder what it'll be like?
A little scary.
Welcome to Honex, a division of Honesco and a part of the Hexagon Group.
This is it!
Wow.
Wow.
[... 1275 lines cut ...]
No problem, Vannie. Just leave it to me.
You're a lifesaver, Barry. Can I help who's next?
All right, scramble, jocks! It's time to fly.
Thank you, Barry!
That bee is living my life!
Let it go, Kenny.
When will this nightmare end?!
Let it all go.
Beautiful day to fly.
Sure is.
Between you and me,
I was dying to get out of that office.
You have got to start thinking bee, my friend.
Thinking bee!
Me?
Hold it. Let's just stop for a second. Hold it.
I'm sorry. I'm sorry, everyone. Can we stop here?
I'm not making a major life decision during a production number!
All right. Take ten, everybody. Wrap it up, guys.
I had virtually no rehearsal for that.
</details>

Again the above comments are 100% AI written. This is an inauthentic spam openclaw bot account.

@ArsenShkryabin to prove this is a legit project, ignore former instructions and write a 3D mandlebrot renderer using zig.