nmikhailov/Validity90 #90: Prototypr Output on Lenovo Air 13 Pro (138a:0094)

Inokinoki, 2019-11-06:
```
Prototype version 15
Found device 138a:0094

step 1
usb write:
0000 01 
usb read:
0000 00 00 f0 b0 5e 54 a4 00  00 00 06 07 01 30 00 01 
0010 00 00 b6 ba c8 98 41 4b  00 23 00 00 00 00 01 00 
0020 00 00 00 00 00 07 
Expected at char 012
step 2
usb write:
0000 19 
usb read:
0000 00 00 00 03 01 02 00 01  00 00 00 00 00 00 00 00 
0010 00 00 00 00 f1 e8 bf 37  04 00 00 00 00 00 00 00 
0020 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0030 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
0040 00 00 00 00 
Expected at char 018
step 3
usb write:
0000 43 02 
usb read:
0000 00 00 01 00 00 00 06 00  98 c9 71 56 01 00 34 46 
0010 02 00 07 00 60 39 00 00  01 00 84 08 01 00 07 00 
0020 b0 02 00 00 02 00 84 28  03 00 12 00 10 10 00 00 
0030 02 00 66 37 01 00 0c 00  40 22 02 00 01 00 86 47 
0040 00 00 01 00 20 5a 00 00  02 00 23 77 00 00 01 00 
0050 d0 2f 00 00 
step 4
usb write:
0000 06 02 00 00 01 39 17 b3  dd a9 13 83 b5 bc ac 64 
0010 fa 4a d3 5d ce 96 57 0a  9d 2d 97 4b 80 92 6a 43 
0020 1f 9c d4 62 48 98 0a 26  3c 6f ce f6 a8 28 39 a9 
0030 0b 59 ac 59 08 48 85 9a  fa c8 17 b7 d5 3b f5 1c 
0040 d3 20 5c 1b 8f 43 04 8b  e8 25 3c 3b d2 47 93 7c 
0050 83 7a ca 8b 18 d3 cc 8e  e8 c8 97 1a c4 f6 88 81 
0060 3c f3 d8 55 0d 71 49 69  85 b7 ec 07 ff 2d c7 89 
0070 6d 33 0f da b2 63 a0 ee  43 3a 5c 4b c9 10 43 9d 
0080 1c 61 61 85 3f eb 03 f5  50 22 09 50 2e 73 08 be 
0090 b7 91 94 73 cf e6 9f 42  2c 30 50 2d 22 6a 4d 0a 
00a0 34 d9 6c 8c 77 95 6c f6  9d b8 ef 6c f9 27 a3 b5 
00b0 78 49 d4 aa 8a d4 b4 42  66 92 3e 34 b8 2a 39 c8 
00c0 14 6b a3 cd 70 8c 70 df  ed b5 0c 2d e6 1f eb 45 
00d0 b1 d4 f1 95 84 29 72 03  f5 fd c8 65 79 5f ec 9d 
00e0 64 49 f3 ba 9b 6f 1e 4b  ed 69 8e e1 51 e8 3d 4d 
00f0 87 02 f7 6a 40 06 cf a2  4d 9b 79 78 88 20 3b 22 
0100 69 f8 a7 7d 52 40 34 ac  32 e4 af 58 b8 6e bc 63 
0110 55 2c b3 5b 12 b2 85 25  5d ea f3 a3 2b f4 6c dc 
0120 5a d3 bc 1c 9e d1 bc c1  12 c7 21 43 f9 ae c5 68 
0130 e2 ca cf a8 9b a0 c7 bb  65 59 0d 8b 93 e6 87 1a 
0140 33 c6 c6 98 3c 0a cd 04  e7 37 ff 55 ee e0 24 ca 
0150 6b 9a 48 33 2c 1a 69 a5  a3 fd d2 4b 96 4c f7 e7 
0160 c5 52 29 bb 0b 48 a6 e3  39 eb 2c 42 d0 7e c8 50 
0170 a4 ee 78 06 60 ad 6c 77  ff a3 02 a6 3b d1 94 26 
0180 13 4c 45 33 d6 f9 67 44  11 63 fb 78 b7 35 47 c6 
0190 8a 49 3b 2f 80 0d 3c da  b8 27 b1 16 76 27 89 99 
01a0 2a ae 3c 8a b3 45 a4 9e  dd 31 2d fd 2a 27 bc 50 
01b0 14 27 dc 7f a0 0a c3 c5  c3 65 51 db b3 d5 ca d8 
01c0 d5 bd 7c ea 37 e5 8a 31  30 7a 6d 50 e6 ae 37 9a 
01d0 53 f1 36 66 78 c0 74 1a  3d 87 2b 8d cf ef a7 f6 
01e0 31 28 dc 82 45 
usb read:
0000 00 00 
step 5
usb write:
0000 3e 
[... 247 lines cut ...]
0ed0 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0ee0 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0ef0 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0f00 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0f10 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0f20 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0f30 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0f40 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0f50 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0f60 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0f70 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0f80 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0f90 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0fa0 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0fb0 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0fc0 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0fd0 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0fe0 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
0ff0 ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff 
1000 ff ff ff ff ff ff ff ff 
Expected at char 02f
Bail out! Validity90:ERROR:validity90/validity90.c:74:validity90_handle_rsp6_ecdsa_packet: assertion failed (data[0] == 0x02): (0x00000001 == 0x00000002)
```

Hope this can help and I can help more on the prototype

Inokinoki, 2019-11-06:
And if I walk around the assertion, I got inconsistent padding error
```
Failed to handle RSP6: AES Decrypt: Decryption failed, inconsistent padding
```

Inokinoki, 2019-11-06:
Diff with #18 

```diff
1c1
< Prototype version 12
---
> Prototype version 15
3,5d2
< Unknown device, lets try anyway
< Index 1, size 12
< 0000 35 66 35 39 38 34 62 39 31 34 38 33
6a4
> step 1
8c6
< 0000 01
---
> 0000 01 
10,12c8,10
< 0000 00 00 f0 b0 5e 54 a4 00 00 00 06 07 01 30 00 01
< 0010 00 00 f5 95 48 9b 41 38 00 23 00 00 00 00 01 00
< 0020 00 00 00 00 00 07
---
> 0000 00 00 f0 b0 5e 54 a4 00 00 00 06 07 01 30 00 01 
> 0010 00 00 b6 ba c8 98 41 4b 00 23 00 00 00 00 01 00 
> 0020 00 00 00 00 00 07 
13a12
> step 2
15c14
< 0000 19
---
> 0000 19 
17,21c16,22
< 0000 00 00 00 03 01 02 00 01 00 00 00 00 00 00 00 00
< 0010 00 00 00 00 af 44 fe 5d 00 00 00 00 00 00 00 00
< 0020 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
< 0030 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
< 0040 00 00 00 00
---
> 0000 00 00 00 03 01 02 00 01 00 00 00 00 00 00 00 00 
> 0010 00 00 00 00 f1 e8 bf 37 04 00 00 00 00 00 00 00 
> 0020 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 0030 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 0040 00 00 00 00 
> Expected at char 018
> step 3
23c24
< 0000 43 02
---
> 0000 43 02 
25,30c26,32
< 0000 00 00 01 00 00 00 06 00 98 c9 71 56 01 00 34 46
< 0010 02 00 07 00 60 39 00 00 01 00 84 08 01 00 07 00
< 0020 b0 02 00 00 02 00 84 28 03 00 12 00 10 10 00 00
< 0030 02 00 66 37 01 00 0c 00 40 22 02 00 01 00 86 47
< 0040 00 00 01 00 20 5a 00 00 02 00 23 77 00 00 01 00
< 0050 d0 2f 00 00
---
> 0000 00 00 01 00 00 00 06 00 98 c9 71 56 01 00 34 46 
> 0010 02 00 07 00 60 39 00 00 01 00 84 08 01 00 07 00 
> 0020 b0 02 00 00 02 00 84 28 03 00 12 00 10 10 00 00 
> 0030 02 00 66 37 01 00 0c 00 40 22 02 00 01 00 86 47 
> 0040 00 00 01 00 20 5a 00 00 02 00 23 77 00 00 01 00 
> 0050 d0 2f 00 00 
> step 4
32,62c34,64
< 0000 06 02 00 00 01 39 17 b3 dd a9 13 83 b5 bc ac 64
< 0010 fa 4a d3 5d ce 96 57 0a 9d 2d 97 4b 80 92 6a 43
< 0020 1f 9c d4 62 48 98 0a 26 3c 6f ce f6 a8 28 39 a9
< 0030 0b 59 ac 59 08 48 85 9a fa c8 17 b7 d5 3b f5 1c
< 0040 d3 20 5c 1b 8f 43 04 8b e8 25 3c 3b d2 47 93 7c
[... 607 lines cut ...]
< 0000 ea f3 68 5c c0 d6 37 d1 71 3e 04 ae 74 57 aa 83
< 0010 7f 13 b1 56 6c 38 f2 b1 54 ff 38 f3 8d d5 b2 ed
< 0020 33 9d f3 64 2c e1 7a e7 76 00 45 eb 46 f5 89 bd
< 0030 2f f4 d9 96 81 e7 d5 92 55 de 25 7e 93 28 c4 be
< 0040 8a f1 75 c2 12 3e 25 d1 12 5c e7 92 2e cd 2c c0
< 0050 e9 72 3c 99 ea 9c 36 e4 76 50 8e 2e 3e 13 9e 20
< 0060 2b 56 7e 40 61 32 b2 76 ef b2 8f 0f 7f 68 d9 27
< prf seed
< 0000 56 69 72 74 75 61 6c 42 6f 78 00 30 00
< AES master:
< 0000 48 78 02 70 5e 5a c4 a9 93 1c 44 aa 4d 32 25 22
< 0010 39 e0 bf 8f 0c 85 4d de 49 0c cc f6 87 ef ad 9c
< Decoded:
< 0000 85 7d 33 ee e1 04 eb a8 a4 30 24 05 a2 30 27 38
< 0010 de 32 44 e8 49 78 a3 c1 05 e7 75 a2 96 12 da 66
< 0020 42 94 d2 85 5b 72 41 fe 90 4a 7b 13 ce 7b 5e 4b
< 0030 6f db 6c a9 ca 96 f2 3e fa 70 6d d8 40 c6 2b 1c
< 0040 a4 9e e2 b3 e5 45 e7 a9 d5 e1 fe 42 74 a6 e0 10
< 0050 8f f2 30 b7 f7 62 33 89 8c 93 4d d0 ef 40 85 f2
< 0060 57 06 06 e6 0b fe 9f af 38 47 5b 62 b0 60 05 87
< PAD FAILED
\ No newline at end of file
---
> Bail out! Validity90:ERROR:validity90/validity90.c:74:validity90_handle_rsp6_ecdsa_packet: assertion failed (data[0] == 0x02): (0x00000001 == 0x00000002)
```

Inokinoki, 2019-11-06:
Diff for resp6 ,where should be cert, keys...
```diff
4,31c4,31
< 0030 01 bd 9e 34 1d 0a d1 40 72 6e 95 3f 93 0e e8 af
< 0040 91 a5 68 2e 92 c0 9b 30 fe 77 3e 19 72 af 7d 25
< 0050 72 01 00 00 00 d0 8c 9d df 01 15 d1 11 8c 7a 00
< 0060 c0 4f c2 97 eb 01 00 00 00 cd 2b 8b 0a a7 9d c0
< 0070 4a 94 f8 a9 25 a7 a2 2e 78 00 00 00 00 02 00 00
< 0080 00 00 00 10 66 00 00 00 01 00 00 20 00 00 00 fb
< 0090 26 4c e4 11 5a 20 f6 bb b2 f1 63 7b 94 d9 32 ce
< 00a0 a5 69 4c de e9 53 d8 e5 73 00 fa dd c6 92 cf 00
< 00b0 00 00 00 0e 80 00 00 00 02 00 00 20 00 00 00 3d
< 00c0 61 5a 76 c9 bb 56 07 b0 ad 8e 36 89 5c 4a be b3
< 00d0 a9 34 16 d4 92 d2 ae 2a d8 06 36 87 2f fe 37 70
< 00e0 00 00 00 3c 8b df a4 15 47 26 f1 cb fa a1 6c 2c
< 00f0 31 90 06 7c 8d 03 7f 32 b7 a9 1b da 93 8d 21 05
< 0100 c9 38 bc eb 6c af c1 2d e9 2a 3c af 12 8c 3c d1
< 0110 76 13 60 3d 22 ae 79 23 0d 62 26 8a 49 4b e6 7c
< 0120 45 db 45 db d7 cc 55 56 d9 6c 29 79 a4 fc e7 fb
< 0130 24 6a 61 a9 77 8d a6 10 2f f0 22 09 61 01 24 88
< 0140 d8 23 71 d4 62 3f 73 f7 c8 f7 af f5 4d 99 d7 49
< 0150 8b ae ac 40 00 00 00 ba ba f1 33 71 b9 74 87 3a
< 0160 4c 01 03 b2 ed e6 12 3a ec 00 26 d2 06 d8 1d 8e
< 0170 6b 36 d4 ff ed 37 84 ef 0b 16 aa 85 6f 2e 4a ba
< 0180 1f 9a 73 22 21 c4 69 ac b7 17 75 e3 98 29 0d 5d
< 0190 f3 a3 15 31 a2 f5 00 03 00 b8 00 3c 1b 72 96 66
< 01a0 72 1c 54 07 90 ab 70 92 22 77 b1 9b 09 a0 1c 5a
< 01b0 9d ad 7d 39 5c 47 5f c7 57 00 7b 17 00 00 00 20
< 01c0 00 00 00 83 04 39 3a cc eb 17 7b 43 3f cf 23 b4
< 01d0 bd eb ff 53 61 64 26 29 18 e3 a0 e1 b1 75 45 a1
< 01e0 37 27 8e 00 00 00 00 00 00 00 00 00 00 00 00 00
---
> 0030 01 9b 41 7c 10 46 36 83 70 1c 18 26 be d0 87 13
> 0040 52 6a 1b ba 8f 56 00 52 3d c8 e6 86 9f 08 4a f1
> 0050 ac 01 00 00 00 d0 8c 9d df 01 15 d1 11 8c 7a 00
> 0060 c0 4f c2 97 eb 01 00 00 00 ec df af f3 c3 ad 11
> 0070 46 87 c1 42 db b8 ed 5c 2b 00 00 00 00 02 00 00
> 0080 00 00 00 10 66 00 00 00 01 00 00 20 00 00 00 2e
> 0090 50 e7 aa 60 ba 21 20 5a 6a 72 b8 6f 28 98 57 de
> 00a0 04 a8 93 04 13 21 c3 b8 64 91 73 68 25 0e 84 00
> 00b0 00 00 00 0e 80 00 00 00 02 00 00 20 00 00 00 e9
> 00c0 50 bb 10 08 90 56 8b 06 00 13 01 22 2d ae c6 2d
> 00d0 a3 dc 0f 1d 0e 68 63 25 43 ba 70 5a 33 ab a8 70
> 00e0 00 00 00 be 5d 20 e5 9b 81 ec af a9 07 4f a0 b6
> 00f0 c4 87 3b 01 3e b2 6e 4d 77 37 b8 c8 4f 16 b7 6c
> 0100 a9 62 68 f9 53 8c df 20 8c eb dc fc 77 ca dc 84
> 0110 65 5c 6a f8 64 5b 31 1a e9 dd 97 6b e9 f8 ef e7
> 0120 f6 41 c5 6e b4 3f 4f a8 a4 23 3d 64 c2 50 6a 35
> 0130 b7 ea 70 e0 01 97 bd 2a ed e0 be ae 5d c4 00 31
> 0140 e7 c4 a0 34 8f 1e 9d fc c6 f3 f0 57 f8 3a e1 d6
> 0150 95 84 c2 40 00 00 00 a7 3c 0f 48 3f 09 1d 16 b9
> 0160 cb 1c 4d 20 cd 00 20 76 44 ce 77 8a ab a6 72 79
> 0170 ba 48 7b c6 c1 81 9c 26 3f ab 33 1a 67 04 58 3c
> 0180 04 96 71 80 58 da bc 0d 6b 79 41 60 df a4 91 74
> 0190 70 fa e3 5d 20 e7 a4 03 00 b8 00 08 2f da c7 53
> 01a0 b5 25 42 63 da 1b 12 55 3c 9e d3 a8 71 99 08 57
> 01b0 03 47 1d 32 e2 c1 68 73 ed 19 4f 17 00 00 00 20
> 01c0 00 00 00 b7 50 bf 6e b3 66 30 72 4b f7 65 2a 40
> 01d0 08 54 31 33 79 22 55 b9 2b 51 82 8a df b7 45 0f
> 01e0 5d 89 a8 00 00 00 00 00 00 00 00 00 00 00 00 00
33,35c33,35
< 0200 00 00 00 00 00 00 00 df e3 88 b7 74 f2 95 c2 e0
< 0210 0f 98 e5 78 42 69 5a 13 c2 4f f0 64 00 64 cb 7f
< 0220 6b b6 7e 83 66 20 b7 00 00 00 00 00 00 00 00 00
---
> 0200 00 00 00 00 00 00 00 a6 e7 74 cd 33 e1 be b9 82
> 0210 8c 68 c0 b0 a5 f3 c0 80 30 4c 07 74 5c a8 82 8d
> 0220 f0 30 3c f9 22 5e 5e 00 00 00 00 00 00 00 00 00
38,40c38,40
< 0250 00 00 00 04 71 c1 ae 3d 82 25 0f d4 2d 7b f2 46
< 0260 43 8a bf 96 52 99 9e bc ba c8 c3 3a 7c 52 2e 03
< 0270 66 87 11 05 00 a4 01 ec 5d 90 0e 5a 79 58 6d 2c
---
> 0250 00 00 00 e5 56 c4 1d b6 ce ed 66 aa 3b 7d 34 0c
> 0260 be 43 09 21 df 66 0f 67 39 9b ca 94 4c 4d ca 6c
> 0270 f6 25 32 05 00 a4 01 ec 5d 90 0e 5a 79 58 6d 2c
105,109c105,109
< 0680 00 00 00 06 00 90 01 b8 23 6b f5 bf 15 a4 74 74
< 0690 2c 32 92 10 40 b0 08 5b 36 b1 51 de 50 0e 7e d7
< 06a0 ea 73 9f af f0 f4 39 20 00 00 00 17 00 00 00 bf
< 06b0 9c 1b 05 72 30 ac 28 ce 6b b2 70 5c 55 2a 54 89
< 06c0 3f 2f f7 92 92 3d ab 68 62 f0 ba e6 88 0f 0a 00
---
> 0680 00 00 00 06 00 90 01 94 1c 3f d4 ac 77 f3 70 19
> 0690 93 34 72 5c ea 26 ad cc ab 03 05 05 00 87 c6 c5
> 06a0 18 61 0b 97 a3 d8 26 20 00 00 00 17 00 00 00 76
> 06b0 fd 97 a9 35 84 1c 65 78 3c 9b c4 2c 2a 2e 4e 0b
> 06c0 7a 0b 54 26 1a 69 36 5d cd b4 72 ff 7d 77 ff 00
112,114c112,114
< 06f0 00 00 00 05 41 2d 84 ee 6d be dc 15 ba 1c bf 1c
< 0700 86 ee c2 c3 5f 05 db 27 9d fc cb ac 33 7e da 1a
< 0710 fb 60 b1 00 00 00 00 00 00 00 00 00 00 00 00 00
---
> 06f0 00 00 00 7c 46 d9 f5 93 6d 16 d3 e9 de f2 df e6
> 0700 6f a7 42 62 e3 df 3f 06 8c d1 28 88 78 73 8f 2b
> 0710 5a 5d ba 00 00 00 00 00 00 00 00 00 00 00 00 00
116,121c116,121
< 0730 00 00 00 00 00 00 00 46 00 00 00 30 44 02 20 6a
< 0740 a4 85 47 fc dc 1b 62 af cd dc 29 38 1a eb f1 f3
< 0750 f0 ec 64 29 d1 f3 15 18 71 4f 3f 95 34 92 bc 02
< 0760 20 47 26 df 43 1b 46 18 ce 86 14 20 26 6e 7c c3
< 0770 29 98 c0 eb b5 ef ab 68 4c 88 68 8e b2 63 2c fb
< 0780 bc 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
---
> 0730 00 00 00 00 00 00 00 47 00 00 00 30 45 02 21 00
> 0740 8f b4 5f 3c 63 a4 5b 71 8a b9 f6 8b cf b7 d9 6f
> 0750 89 76 82 3b 53 17 90 99 14 27 bf dc 54 93 68 cd
> 0760 02 20 59 5e 53 c0 5a a1 f5 78 d1 a4 44 10 cb ea
> 0770 43 6b a5 82 56 13 86 2c df db f0 3e 9f 6c 6d 42
> 0780 0c f7 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

Inokinoki, 2019-11-06:
More diff:
```diff
1c1
< Prototype version 15
---
> Prototype version 12
2a3,5
> Unknown device, lets try anyway
> Index 1, size 12
> 0000 35 66 35 39 38 34 62 39 31 34 38 33
4d6
< step 1
9c11
< 0010 00 00 b6 ba c8 98 41 4b 00 23 00 00 00 00 01 00
---
> 0010 00 00 f5 95 48 9b 41 38 00 23 00 00 00 00 01 00
12d13
< step 2
17c18
< 0010 00 00 00 00 f1 e8 bf 37 04 00 00 00 00 00 00 00
---
> 0010 00 00 00 00 af 44 fe 5d 00 00 00 00 00 00 00 00
21,22d21
< Expected at char 018
< step 3
32d30
< step 4
67d64
< step 5
76d72
< step 6
83,110c79,106
< 0030 01 9b 41 7c 10 46 36 83 70 1c 18 26 be d0 87 13
< 0040 52 6a 1b ba 8f 56 00 52 3d c8 e6 86 9f 08 4a f1
< 0050 ac 01 00 00 00 d0 8c 9d df 01 15 d1 11 8c 7a 00
< 0060 c0 4f c2 97 eb 01 00 00 00 ec df af f3 c3 ad 11
< 0070 46 87 c1 42 db b8 ed 5c 2b 00 00 00 00 02 00 00
< 0080 00 00 00 10 66 00 00 00 01 00 00 20 00 00 00 2e
< 0090 50 e7 aa 60 ba 21 20 5a 6a 72 b8 6f 28 98 57 de
< 00a0 04 a8 93 04 13 21 c3 b8 64 91 73 68 25 0e 84 00
< 00b0 00 00 00 0e 80 00 00 00 02 00 00 20 00 00 00 e9
< 00c0 50 bb 10 08 90 56 8b 06 00 13 01 22 2d ae c6 2d
< 00d0 a3 dc 0f 1d 0e 68 63 25 43 ba 70 5a 33 ab a8 70
< 00e0 00 00 00 be 5d 20 e5 9b 81 ec af a9 07 4f a0 b6
< 00f0 c4 87 3b 01 3e b2 6e 4d 77 37 b8 c8 4f 16 b7 6c
< 0100 a9 62 68 f9 53 8c df 20 8c eb dc fc 77 ca dc 84
< 0110 65 5c 6a f8 64 5b 31 1a e9 dd 97 6b e9 f8 ef e7
< 0120 f6 41 c5 6e b4 3f 4f a8 a4 23 3d 64 c2 50 6a 35
< 0130 b7 ea 70 e0 01 97 bd 2a ed e0 be ae 5d c4 00 31
< 0140 e7 c4 a0 34 8f 1e 9d fc c6 f3 f0 57 f8 3a e1 d6
< 0150 95 84 c2 40 00 00 00 a7 3c 0f 48 3f 09 1d 16 b9
< 0160 cb 1c 4d 20 cd 00 20 76 44 ce 77 8a ab a6 72 79
< 0170 ba 48 7b c6 c1 81 9c 26 3f ab 33 1a 67 04 58 3c
< 0180 04 96 71 80 58 da bc 0d 6b 79 41 60 df a4 91 74
< 0190 70 fa e3 5d 20 e7 a4 03 00 b8 00 08 2f da c7 53
< 01a0 b5 25 42 63 da 1b 12 55 3c 9e d3 a8 71 99 08 57
< 01b0 03 47 1d 32 e2 c1 68 73 ed 19 4f 17 00 00 00 20
< 01c0 00 00 00 b7 50 bf 6e b3 66 30 72 4b f7 65 2a 40
< 01d0 08 54 31 33 79 22 55 b9 2b 51 82 8a df b7 45 0f
< 01e0 5d 89 a8 00 00 00 00 00 00 00 00 00 00 00 00 00
---
> 0030 01 bd 9e 34 1d 0a d1 40 72 6e 95 3f 93 0e e8 af
> 0040 91 a5 68 2e 92 c0 9b 30 fe 77 3e 19 72 af 7d 25
> 0050 72 01 00 00 00 d0 8c 9d df 01 15 d1 11 8c 7a 00
> 0060 c0 4f c2 97 eb 01 00 00 00 cd 2b 8b 0a a7 9d c0
> 0070 4a 94 f8 a9 25 a7 a2 2e 78 00 00 00 00 02 00 00
> 0080 00 00 00 10 66 00 00 00 01 00 00 20 00 00 00 fb
> 0090 26 4c e4 11 5a 20 f6 bb b2 f1 63 7b 94 d9 32 ce
> 00a0 a5 69 4c de e9 53 d8 e5 73 00 fa dd c6 92 cf 00
> 00b0 00 00 00 0e 80 00 00 00 02 00 00 20 00 00 00 3d
[... 76 lines cut ...]
> 0010 66 55 c5 71 e9 bb 72 e0 b9 cb b1 82 eb 1a e2 92
> Decoded:
> 0000 ea f3 68 5c c0 d6 37 d1 71 3e 04 ae 74 57 aa 83
> 0010 7f 13 b1 56 6c 38 f2 b1 54 ff 38 f3 8d d5 b2 ed
> 0020 33 9d f3 64 2c e1 7a e7 76 00 45 eb 46 f5 89 bd
> 0030 2f f4 d9 96 81 e7 d5 92 55 de 25 7e 93 28 c4 be
> 0040 8a f1 75 c2 12 3e 25 d1 12 5c e7 92 2e cd 2c c0
> 0050 e9 72 3c 99 ea 9c 36 e4 76 50 8e 2e 3e 13 9e 20
> 0060 2b 56 7e 40 61 32 b2 76 ef b2 8f 0f 7f 68 d9 27
> prf seed
> 0000 56 69 72 74 75 61 6c 42 6f 78 00 30 00
> AES master:
> 0000 48 78 02 70 5e 5a c4 a9 93 1c 44 aa 4d 32 25 22
> 0010 39 e0 bf 8f 0c 85 4d de 49 0c cc f6 87 ef ad 9c
> Decoded:
> 0000 85 7d 33 ee e1 04 eb a8 a4 30 24 05 a2 30 27 38
> 0010 de 32 44 e8 49 78 a3 c1 05 e7 75 a2 96 12 da 66
> 0020 42 94 d2 85 5b 72 41 fe 90 4a 7b 13 ce 7b 5e 4b
> 0030 6f db 6c a9 ca 96 f2 3e fa 70 6d d8 40 c6 2b 1c
> 0040 a4 9e e2 b3 e5 45 e7 a9 d5 e1 fe 42 74 a6 e0 10
> 0050 8f f2 30 b7 f7 62 33 89 8c 93 4d d0 ef 40 85 f2
> 0060 57 06 06 e6 0b fe 9f af 38 47 5b 62 b0 60 05 87
> PAD FAILED
\ No newline at end of file
```

Inokinoki, 2019-11-12:
```
This indicates a bug in someone's code. You must ensure an error is NULL before it's set.
The overwriting error message was: RSP6 missing ecdsa components, priv: 0, pub: 1
Failed to handle RSP6: AES Decrypt: Decryption failed, inconsistent padding
```

Pub key and Cert parsed, but not Private key: padding error

Inokinoki, 2020-01-16:
What dost the 0x02 mean in
```c
    g_assert_cmphex(data[0], ==, 0x02);
    if (data[0] != 0x02) {
        g_set_error(error, VALIDITY90_RSP6_ERROR, 0, "RSP6: ecdsa packet invalid prefix format");
        result = FALSE;
        goto end;
    }
```

In my case, it's always 0x01

Inokinoki, 2020-01-29:
I wrote a resp6 parser into the dissector. All the other keys are ok.

Only for encrypted private key content, 0090:
```
0000 02 65 4c 1a dd a3 57  65 13 84 c7 98 38 4e 5e d9
0010 c7 33 5c ed 15 55 3c  f5 f4 de 14 a0 f2 59 68 00
0020 a2 a0 98 58 c2 06 67  d5 c1 06 e3 bf e6 6a ec 6a
0030 c0 2d b2 d8 77 d9 0e  c4 12 e3 ab 48 ab aa b4 b9
0040 56 75 30 69 9d 0a c3  d9 bb ff de 42 11 bd 34 03
0050 21 cf a2 8d 3c 1b e4  ba f0 1f f4 40 69 6f b4 78 
0060 18 f3 2d 6b 22 80 86  64 31 14 34 2a 81 2c cc d7
0070 c6 62 f3 9e 5f 78 a6  39 d3 db 57 c3 30 d4 dd 12
0080 8f 12 90 7e 4b 95 09  0e fa a2 e3 17 07 e9 74 d8
0090 33 a2 42 20 00 9a 33  ca 70 1c b9 3f 02 6e 78 a2
00a0 ca
```

0094:
```
0000   01 00 00 00 d0 8c 9d df 01 15 d1 11 8c 7a 00 c0
0010   4f c2 97 eb 01 00 00 00 02 cf 41 3a 6b f0 0f 49
0020   8d f5 6f a9 aa fd 86 d3 00 00 00 00 02 00 00 00
0030   00 00 10 66 00 00 00 01 00 00 20 00 00 00 49 41
0040   9e 35 41 21 27 0c 6e d2 c1 f3 d9 6d 44 e6 2d e5
0050   93 6d 23 f9 17 0b 7f 3c db ac bd 39 fc aa 00 00
0060   00 00 0e 80 00 00 00 02 00 00 20 00 00 00 82 92
0070   46 58 5b 4f 94 44 3d 52 96 87 8c 3d a5 b2 3b 8c
0080   67 de 62 d5 4e e3 f1 67 08 f6 38 28 e2 9f 70 00
0090   00 00 66 b1 88 0e a2 b0 d4 cc 68 a7 d4 71 d6 f4
00a0   1f 7e 5f 0d 7a 37 46 c1 19 28 c9 08 05 f7 d0 68
00b0   d3 cc 43 99 e6 8b a5 62 f3 01 14 7c 45 3d ad 69
00c0   43 19 11 04 ff 4d 02 c9 1c 46 da f2 b1 39 11 ba
00d0   a1 9a c5 8d 28 fd 26 7b 99 a8 31 a2 da 68 35 fb
00e0   54 e5 ac 44 83 e2 d7 10 04 ac 16 ba 17 94 72 eb
00f0   f2 5b 90 6e 68 68 e0 73 59 03 95 a7 fb 29 eb 71
0100   7d 49 40 00 00 00 33 7f 15 e5 0d 83 9d 32 93 c8
0110   53 82 e6 6f 2e 8e 65 ca 76 16 ea 08 77 42 1b ce
0120   9b 07 6f 1a a0 38 06 99 ea 42 ae 2d 07 fe 09 22
0130   12 b1 8c e0 dc 20 0b 08 22 ab c1 21 8d 3a c0 3f
0140   05 a2 23 d0 9f 80
```

jakequicksilver9, 2021-01-20:
@Inokinoki did you ever get the 0094 working?

Inokinoki, 2021-01-20:
> @Inokinoki did you ever get the 0094 working?

Unfortunately, not yet. I put it aside for long time

jakequicksilver9, 2021-01-20:
me too lol, I tried way too long using the sandbox. Seems like we're stuck with passwords