nmikhailov/Validity90 #94: Please note that nmikailov has not been active since Feb 2019.

jontis, 2020-01-06:
Please note that nmikailov has not been active since Feb 2019.

MartinX3, 2020-01-06:
Does that mean no one will upstream the changes?

jontis, 2020-01-07:
Sadly, I believe so.
This project has huge interest and a lot of people seems to be willing to put in an effort for it but something seems to have happened with the repo owner.
I hope the repo owner may appoint a deputy to share the administration or that someone would fork it and continue the work.

maciejmotyka, 2020-02-16:
I used https://github.com/hbbio/forkizard to check which forks are ahead and found 3, but they don't seem to have achieved any meaningful progress.

```
./forkizard nmikhailov/Validity90
2020/02/16 09:47:57 103 forks

/ripper2hl/Validity90       +2 -0
/globotree/Validity90       +1 -0
/imdark/Validity90          +1 -0
```

I will donate if anyone picks up the work for 06cb:009a.

MartinX3, 2020-07-03:
https://github.com/3v1n0/libfprint

Here is the successor.

pianist, 2021-08-03:
Anybody tried to contact Nikita?