nmikhailov/Validity90 #100: Prototype output on Lenovo V330-14ARR (06cb:0081)

lozanotux, 2020-02-29:
OS: Fedora 31 Workstation (amd64)
Kernel: 5.5.6-201
Prototype output:

```
juan@V330-14ARR:~/Downloads/Validity90/prototype$ ./prototype --help
Prototype version 15
Found device 06cb:0081
Support would be available soon
```
And try with **Fingerprint GUI** but without any results:
[image]

hackrx, 2020-08-26:
Is there any update for this?
I am using Lenovo Ideapad 520, and this is my hw.info of fingerprint
```
usb:06cb-0081;ff-00-00;failed;fingerprint reader;;Synaptics, Inc.;
```

raaz2053, 2025-05-01:
Could you please provide an update on the status of the fingerprint reader issue? I'm using a Lenovo  Yoga 730