nmikhailov/Validity90 #104: Ubuntu 20.04, new libfprint-2 and as such our precious drivers won't work.

spxak1, 2020-05-03:
Nice easy compile and installation in 19.10

Upgraded to 20.04, the fingerprint scanner stopped working. 

After looking at what changed, libfprint is now libfprint-2, installs a new set of libraries (libfprint-2.so.2.0.0) at a different location (/usr/lib/x86_64-linux-gnu).

Our drivers compile great, but they're no substitute for the new libraries, as such I'm currently back to square one.

This is on 138a:0090. Thank you

4khobta, 2020-05-06:
138a:0097 - the same

MartinX3, 2020-05-20:
I have the sad/slowly feeling this wonderful project is dead. :(

suiahae, 2020-05-29:
Fedora 32 updated fprintd to the latest version, which also requires libfprint-2.so. The previous modified version is completely unusable. Could successfully make and install it, and run the enroll and verify in the examples directory. But running fprintd-enroll gave No devices available.

ThinkPad T470 with 138a:0097, too.

MartinX3, 2020-07-03:
https://github.com/3v1n0/libfprint

Here is the successor.

suiahae, 2020-07-04:
> https://github.com/3v1n0/libfprint
> 
> Here is the successor.

It cannot be used in Fedora 32. When I tested `meson libfprint libfprint/_build`, I got this error following:
[image]

MartinX3, 2020-07-04:
Does the PKGBUILD from https://aur.archlinux.org/packages/libfprint-vfs009x-git/ helping you?
(Just checkout the `Git Clone URL:`)

Or how it get build there for ubuntu?
https://launchpad.net/~3v1n0/+archive/ubuntu/libfprint-vfs0090/+packages

I am sadly not much into fedora, but maybe it's just a configuration mistake on your build system?

sircubbi, 2020-07-04:
Well, if it helps I uploaded by private builds of libfprint for Fedora 32 (64 bit) here: http://cubbi.de/libfprint/

suiahae, 2020-07-04:
@MartinX3 @sircubbi thanks for your help! 
Through "PKGBUILD of the arch repository" and "@sircubbi private builds of libfprint", I successfully installed the new libfprint. **However**, when I used fprintd-enroll to test, I found that they didn't work ;;;;
[image]

tgelter, 2020-07-04:
> @MartinX3 @sircubbi thanks for your help!
> Through "PKGBUILD of the arch repository" and "@sircubbi private builds of libfprint", I successfully installed the new libfprint. **However**, when I used fprintd-enroll to test, I found that they didn't work ;;;;
> [image]

This is exactly what I ran into as well: https://gitter.im/Validity90/Lobby?at=5efe3170e0e5673398ebc0be

MartinX3, 2020-08-01:
https://github.com/uunicorn/python-validity
For new users