nmikhailov/Validity90 #107: Final solution to 009a

sequencer, 2020-05-20:
FYI.
x1c6 and x1c7 has a same mold of finger print device, since fingerprint device is under USB, which has no whitelist restriction. So just buy a `06cb:00bd` in ebay or taobao (~10$) and replace the original one. And we can finally archive this repository!!!

tester1969, 2020-05-23:
Can't find anything, if you can give a link or something

sequencer, 2020-05-23:
For example: 
https://item.taobao.com/item.htm?spm=a1z09.2.0.0.32f62e8dA6XAVL&id=614039278086&_u=n565t2062ea
https://www.ebay.com/itm/Original-Lenovo-ThinkPad-X1-Carbon-7th-Gen-X1-Extreme-2nd-Fingerprint-Reader-FPR/174135587770?hash=item288b4a33ba:g:VBwAAOSwsmdeAdkF

dennym, 2020-05-23:
But what about all the non X1C thinkpads? Like T480?

maciejmotyka, 2020-05-25:
According to the seller's description, the device that @sequencer linked to should fit
>T590, X390, X395, T490, T495, T490S, T495S, X390 YOGA, P53, P53S, P43S, X1 Carbon 7th, X1 Extreme 2nd

This offer
https://www.ebay.com/itm/original-for-lenovo-for-THINKPAD-T480-T480S-X280-A285-E480-Fingerprint-Reader/283584355427?hash=item4206f1dc63:g:RtkAAOSwf1VdWWVz
is for a sensor that supposedly fits
> T480 T480S X280 A285 E480

Upon brief visual inspection, these two sensors look identical with regards to their shape and position of the connector. So there's a possibility that you can use the same sensor in all of these laptops.

djmandell, 2020-06-07:
I decided to bite the bullet and try the swap.  I bought one of the 06cb:00bd devices from eBay (you can search for the FRU 5F30V25898 or the chip codename Prometheus).  I ended up buying this one: https://www.ebay.com/itm/5F30V25898-FRU-Prometheus-BK-JYT-CARDS-MISC-INTERNAL-FP-READER/174197141376?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649 

I've got good news and bad news about this chip.  I'll start with the good news first.  The chip is the right size, the connector's in the right place, and it works perfectly in Windows and Ubuntu 20.04.  In Linux I can enroll fingerprints and authenticate using the fingerprint reader.  If you're really itching to have a functional fingerprint reader in Linux, this thing will do the job.

Now the bad news.  First, I've got a t480 and pulling the chip out was a REAL pain.  In this particular model you have to pull the motherboard out before you can connect to the chip, which means disconnecting everything and putting it all back together again.  I accidentally broke the little black clasp on the touchpad connector, so that's effectively broken now (thankfully I disabled the touchpad anyway, so that doesn't hurt too bad).  This might not be as much of an issue for other laptop models, or it might not be an issue for you if you're less of a klutz than I am.

Second, while the reader does fit in the the old slot, it is not an exact replacement.  For one thing, this version of the fingerprint reader does not have a status LED built into it (and I guess the t490, t495, etc don't have it either), which means that there's now a little gap where the plastic from the old reader used to be.  I'll probably just cover that with tape.  There are several eBay resellers that sell a chip that does have the LED built-in, but my suspicion is that they're showing a picture of the old chip (the FRU number on printed on the sticker matches the FRU for the old reader).

Also, I'm not 100% sure about this, but I think the black reader part of the fingerprint reader (where you rest your finger) isn't quite as tall as the old one was, the new one seems a bit more recessed.  I'm fully willing to admit that this might all be in my head.

In conclusion, this does work, but if I knew then what I know now I'd probably have just kept the old one and dealt with it not working in Linux.  If you're excited to do this, by all means go for it!  You'll have a functioning fingerprint reader by the time you get to the end.

sequencer, 2020-06-07:
Emmm, as for the LED, I found it reserves the LED place, so as for the LED, I think it more likely that we have no currect driver.

tester1969, 2020-06-07:
@djmandell Thank you for the post. I have t480 too. Man, when you do such things, always get a video :) I ll give it a try too, Although I will need a couple of months to source it from China.

javierfileiv, 2020-07-05:
I think that' s more a workaround than a solution.... Here [python-validity](https://github.com/uunicorn/python-validity/issues/3) are making really good progress. Even though is in python, it can be ported to C I guess

spookius, 2020-07-08:
> I think that' s more a workaround than a solution.... Here [python-validity](https://github.com/uunicorn/python-validity/issues/3) are making really good progress. Even though is in python, it can be ported to C I guess

I managed to get my P52 reader working using python-validity and [following these steps](https://github.com/uunicorn/python-validity/issues/3#issuecomment-655477344)

tester1969, 2020-07-08:
https://github.com/tester1969/pam-validity

danielktaylor, 2020-07-16:
I just did the swap on my t480s.

I ordered from Taobao by using engtb.com and the part arrived in about a month. The cost savings are lower versus ordering from eBay now that epacket rates were increased this month.

Tips for doing the swap:

- You only need to remove the bottom plastic cover and the battery which are both very easy (https://www.youtube.com/watch?v=UHBuyu_Dy3k&feature=youtu.be)
- The screw for the fingerprint reader bracket has loctite on it, so it is hard to remove. Apply pressure from the back while you unscrew it (don't press down on the screen) and it will eventually loosen
- To remove the cable, carefully lift open the black tab
- Make sure you enable the fingerprint reader on the security page of the BIOS
- I needed to to run "sudo pam-auth-update" to enable fingerprint support in Pop!_OS 20.04

Unfortunately, the Taobao seller sent me an older incompatible version of the fingerprint reader, which explains why I needed to drill new mount holes to install it :laughing:

This time I'm ordering from eBay...

closed 2020-11-09 by sequencer