nmikhailov/Validity90 #108: TO ANYONE LOOKING FOR UPDATES ON THE PROJECT

josh-palan, 2020-12-02:
I waited and watched this repository for my than 3 years waiting on support for my Lenovo. I still see people landing on this project asking for support for their reader. Unfortunately, it appears this repo is in a dead state and has been for some time now. :/

That said, I was able to install and use the fprint drivers from the following project:
https://github.com/uunicorn/python-validity

That repo works for me using a Lenovo T580, synaptics device ID 06cb:009a, on Ubuntu 18.04 and 20.04 using the plasma desktop. I would highly suggest that anyone who hasn't checked it out yet heads over there to see if it'll work for them as well, especially since the project seems very much alive.

MartinX3, 2020-12-02:
And the T460p on arch and ubuntu
But he other hardware fprinter should work there, too
(Hoping for upstreaming into the vanilla driver)

bmorgan365, 2020-12-02:
Thank you so much for this, I was starting to think I'd never be able to use the fingerprint sensor!!!

nikeee, 2020-12-02:
Thanks. Just worked on my X1C6 with `06cb:009a` on stock Ubuntu 20.04 (gnome desktop) with dual-boot windows.

jaxjexjox, 2021-01-12:
Thanks for the update, this entry seems to be the most common Google hit still.

mareikep, 2021-01-12:
Not all heroes wear capes... Thanks for the update. Works like a charm on T480s with 06cb:009a

parkerlreed, 2021-01-12:
0094 is still SOL

jaxjexjox, 2021-01-12:
What about 0088?

The other project is having issues right now and you can't add the repository anyhow!

jmaguero, 2021-01-15:
This is amazing! Thanks for the update. Anyone knows how to install it on debian unstable? Thanks!

burdandrei, 2021-01-26:
Thank you very  much!!! Worked on T480 like a charm!

Artionet, 2021-05-29:
Thank you so much! I now have a working fingerprint reader on X1 Yoga 3rd Gen.

Lapazlane, 2022-06-02:
My Thinkpad Yoga 370 works using https://github.com/uunicorn/python-validity on Ubuntu (and POP OS) 20.04, and POP OS 21.10. Sadly not yet working on 22.04 - I (and others) have asked the developer if he will be updating for 22.04.

Lapazlane, 2022-06-02:
> 

The developer has just advised me that he will be updating to 22.04, but is first working on issue #127.

natcore-tech, 2025-11-14:
Muchas gracias, justo necesitaba todo esto.